1technophile/OpenMQTTGateway #232: Bad way to install Arduino ESP32 environment archive

xelfe, 2018-07-11:
The archive [Arduino ESP32 environment](https://github.com/espressif/arduino-esp32/archive/a153f6ad280e5932be9c80341556348ff505d586.zip) cannot work because the BLE libraries is a git submodule and it's empty by default. We cannot make a "git submodule update --init --recursive" because git archive does not contain git info.

There's two fix possible;

1. Doing a git clone with a checkout to the right revision and doing a "git submodule update --init --recursive" to get submodules.
2. Make our own archive that contain submodules.

1technophile, 2018-07-11:
In fact you should not use the BLE library included with ESP32 env but the [one supplied](https://github.com/1technophile/OpenMQTTGateway/tree/master/lib/ESP32_BLE_Arduino) in lib folder of the OpenMQTTGateway repo as indicated [there](https://github.com/1technophile/OpenMQTTGateway/wiki/ESP32-BT-Module)

1technophile, 2018-07-28:
@xelfe did you checked with the library pointed?
If ok for you could you close the issue please, if not feel free to give your returns.

1technophile, 2018-08-05:
Could you take a look to this : 
https://community.openmqttgateway.com/t/esp32-arduino-stable-release-v1/78

1technophile, 2018-09-07:
wiki has been updated with the last working env , more details there:
https://community.openmqttgateway.com/t/ble-and-esp32-dev-environment/102/2

closed 2018-09-07 by 1technophile