1technophile/OpenMQTTGateway #234: Support for IR controller shield (wemos d1 mini)

rradar, 2018-07-21:
Wemos D1 mini IR controller shield specifications:

- **4x IR LEDs** (940nm) pointing in four different directions
- 1x IR receiver (38 kHz) for learning function
- Tx/Rx mode selection – Sender – D3/GPIO0, Receiver – D4/GPIO2
- 2x 8-pin headers compatible with Wemos/LOLIN D1 mini board

[image]
[image]

US $1.40 / piece

Is it for good? Should make a compact desgin in the end :smiling_imp:

rradar, 2018-07-22:
More technical details here: https://wiki.wemos.cc/products:d1_mini_shields:ir_controller_shield

Including schematic's and datasheet's

> Schematic V1.0.0
> IR12-21c Datasheet
> IRM-H638T Datasheet

1technophile, 2018-07-22:
Ordered, will see in a few weeks

rradar, 2018-08-19:
I have it! Soldered and ready to fly! 

Pins are clear:

D3(GPIO0) | Send
D4(GPIO2) | Receiver

Any tips for code to try? Should it just works out of the box?

1technophile, 2018-08-19:
great, just follow the [wiki ](https://github.com/1technophile/OpenMQTTGateway/wiki/ESP8266-IR-Send-and-Receive) on we say to uncomment ZgatewayIR in user_config.h and change the pins in config_IR.h

1technophile, 2018-08-22:
I received mine, don't still tested them, do you have some feedback on your side?

rradar, 2018-08-28:
hey @1technophile 
sorry - didn't had any time recently and had no running arduino sdk as well...
Maybe the next weekend I have time to try - meanwhile you could give a try and report back ;)

1technophile, 2018-09-07:
@rradar I confirm it works natively with these pins:
```
#ifdef ESP8266
  #define IR_RECEIVER_PIN 2
  #define IR_EMITTER_PIN 0
```

Receiver and emitter

Range of emitter is around 2meters for my tv

closed 2018-09-07 by 1technophile