1technophile/OpenMQTTGateway #236: Esp8266 bin

lscosta90br, 2018-07-22:
I would like to configure OpenMQTTGateway to work with esp8266, would anyone have the immediate scope for the tasmota project?

Before submitting a problem please check the troubleshooting section
https://github.com/1technophile/OpenMQTTGateway/wiki/Troubleshooting

So as to have a more efficient treatment of your request please provide the following information:
* OpenMQTTGateway version used (V0.4, 0.3, master...)
* Library version related to the problem you have (if you have troubles with RF provide the version of RCSwitch library)
* For IR and RF clarify if you tested with the basic examples given with these libraries
* You should not have a compilation error if you use the versions of the libraries linked into the libraries folder, this badges show you the state of the compilation 
[image]
* If you are not sure this is a bug or an enhancement post your question to the forum below
[image]

1technophile, 2018-07-23:
you mean the bin file ?

They are not included currently with 0.7 but will be delivered with 0.8

lscosta90br, 2018-07-23:
Very good!!

-- 
[ ]s,
LRU #391067
Analista de Rede
Leandro Costa
:wq
http://www.centosbr.org
...... __@
...._ \ >_
...(_)/ (_)_________

closed 2018-07-28 by 1technophile