1technophile/OpenMQTTGateway #237: Sensors RF Door DW1

lscosta90br, 2018-07-23:
I was able to install my OpenMQttGateway on a esp8266.
I'm using a Sonoff DW1 sensor, how do I know how much the device is open or closed via mqtt.
~
home / OpenMQTTGateway / 433toMQTT / protocol 1
home / OpenMQTTGateway / 433toMQTT / bits 24
home / OpenMQTTGateway / 433toMQTT / length 407
home / OpenMQTTGateway / 433toMQTT 5000000

home / OpenMQTTGateway / 433toMQTT / protocol 1
home / OpenMQTTGateway / 433toMQTT / bits 24
home / OpenMQTTGateway / 433toMQTT / length 406
home / OpenMQTTGateway / 433toMQTT 5000000

I'll use it to control an opening and closing garage!

Before submitting a problem please check the troubleshooting section
https://github.com/1technophile/OpenMQTTGateway/wiki/Troubleshooting

So as to have a more efficient treatment of your request please provide the following information:
* OpenMQTTGateway version used (V0.4, 0.3, master...)
* Library version related to the problem you have (if you have troubles with RF provide the version of RCSwitch library)
* For IR and RF clarify if you tested with the basic examples given with these libraries
* You should not have a compilation error if you use the versions of the libraries linked into the libraries folder, this badges show you the state of the compilation 
[image]
* If you are not sure this is a bug or an enhancement post your question to the forum below
[image]

1technophile, 2018-07-23:
Hello,

The DW1 only send one code when the magnet leaves the sensor.
It is not possible to know if  your garage door is open or close with this sensor. You will only know that it moved.
What I've done for a garage door sensor is to put an sonar [hc-sr04 to detect if the door is in front of the sonar or not ](https://1technophile.blogspot.com/2016/08/garage-door-sensor-arduino-nano.html)

You can maybe find 433mhz door sensor with open and close state also, take a look to[ the compatible devices](https://docs.google.com/spreadsheets/d/1DdtVtSsN25nwP6BZI5q6C9yDGz37tUWjw2SQ1RGwBxU/edit#gid=0)

1technophile, 2018-07-28:
I'm going to close this issue as it is not directly related to OMG, if you want to continue the discussion don't hesitate to open a topic in the [image]

closed 2018-07-28 by 1technophile