1technophile/OpenMQTTGateway #239: Run on Particle Photon?

mixpc, 2018-07-25:
Hello,

This is a great project indeed. Is it possible to use [Particle Photon](https://docs.particle.io/datasheets/photon-datasheet/) to run OpenMQTTGateway? I do not up to what point code for ESP8266 can be re-used in Particle Photon, though, a part from pin numbering, of course. The idea is using the Photon instead of a Raspberry Pi in oder to read/send mqtt messages to/from RF433 sensors and switches.

Thank you in advance,

1technophile, 2018-07-25:
Hello,

I didn't heard about a user having tested this board with OMG but if you want to be the first don't hesitate to do the test. In my side I don't know this platform.

Regards

1technophile, 2018-07-28:
I'm going to close this issue, if you want to continue the discussion with others don't hesitate to open a topic in the [image]

closed 2018-07-28 by 1technophile