1technophile/OpenMQTTGateway #240: Config example miflora home assistant

thundergreen, 2018-07-26:
This is a great project but it is not really clear how to use ubeacon and miflora... I dunno how to set it up properly.. maybe an example would be great.

1technophile, 2018-07-28:
I have examples with OpenHAB but not with home assistant maybe others can help?

mredone, 2018-08-26:
@thundergreen  
Show me yours mqtt topics from beacon and miflora.

A will try make home assistant configuration for you.

oakbrad, 2018-08-29:
Here's an example

** sensors.yaml **
```
- platform: mqtt
  name: kitchen_pothos_temperature
  unit_of_measurement: "°F"
  force_update: true
  expire_after: 21600 # 6 hours
  state_topic: "home/BTtoMQTT/C47C8D6273BD/tem"
  value_template: "{{ (((value | float) * 9 / 5) + 32) | round(1) }}"
- platform: mqtt
  name: kitchen_pothos_moisture
  force_update: true
  expire_after: 21600 # 6 hours
  state_topic: "home/BTtoMQTT/C47C8D6273BD/hum"
- platform: mqtt
  name: kitchen_pothos_light_intensity
  unit_of_measurement: lux
  force_update: true
  expire_after: 21600 # 6 hours
  state_topic: "home/BTtoMQTT/C47C8D6273BD/lux"
- platform: mqtt
  name: kitchen_pothos_conductivity
  force_update: true
  expire_after: 21600 # 6 hours
  state_topic: "home/BTtoMQTT/C47C8D6273BD/fer"
```

```
plant:
  kitchen_pothos:
    sensors:
      moisture: sensor.kitchen_pothos_moisture
      temperature: sensor.kitchen_pothos_temperature
      conductivity: sensor.kitchen_pothos_conductivity
      brightness: sensor.kitchen_pothos_light_intensity
    min_moisture: 10
    max_moisture: 100
    min_conductivity: 200
    min_temperature: 45
```

1technophile, 2018-08-30:
Great, maybe a candidate to enrich the [wiki](https://github.com/1technophile/OpenMQTTGateway/wiki/Home-Assistant-integration)?

Destix, 2018-09-05:
As it's for wiki add this line

**configuration.yaml**

`sensor: !include sensors.yaml`

oakbrad, 2018-09-05:
Wiki has been updated

1technophile, 2018-09-05:
@thundergreen if ok for you, could you close the issue please

closed 2018-09-07 by 1technophile