1technophile/OpenMQTTGateway #241: ESP32 crashing using BT and RF modules

clyra, 2018-07-27:
I'm using version 0.8beta and version 0.7. Both versions seems to be stable enough while using only BT feature, but after enabling RF I got a lot of reboots and even total freezes with messages like the one below:

Guru Meditation Error: Core  1 panic'ed (Cache disabled but cached memory region accessed)
Core 1 register dump:
PC      : 0x401a1b44  PS      : 0x00060034  A0      : 0x80080f96  A1      : 0x3ffc0ba0  
A2      : 0x3ffc5204  A3      : 0x00000004  A4      : 0x3ffc1124  A5      : 0x000000ff  
A6      : 0x3ffc52a0  A7      : 0x00000000  A8      : 0x800812cd  A9      : 0x00000001  
A10     : 0x00000004  A11     : 0x00000000  A12     : 0x800901a5  A13     : 0x3ffd9da0  
A14     : 0x00000001  A15     : 0x00000000  SAR     : 0x0000001e  EXCCAUSE: 0x00000007  
EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000  
Core 1 was running in ISR context:
EPC1    : 0x40082404  EPC2    : 0x00000000  EPC3    : 0x00000000  EPC4    : 0x401a1b44

Backtrace: 0x401a1b44:0x3ffc0ba0 0x40080f93:0x3ffc0bc0 0x40081041:0x3ffc0bf0 0x40081861:0x3ffc0c10 0x40082401:0x00000000

If I just disconnect the RF module (pull the data pin cable of the RF receiver) it resume the normal operation. Since this problem usually happens before the first complete BT scan and if you boot with the RF pin unplugged and plug it later it works, my guess is that something is happening in the setup/init phase of BT.

1technophile, 2018-07-28:
Thanks for this investigation, could you give me the pins you are using for RF and what is the revision of ESP32 env you are using?

clyra, 2018-07-28:
Hi,

Arduino IDE 1.8.5. Esp32 env and other libs as per
https://github.com/1technophile/OpenMQTTGateway/wiki/Module-Overview-ESP32
instructions. I couldn't get the RF working with the default pin but get it
right with the "alternative" pin 27.

clyra, 2018-08-03:
Hi,
It seems that I had a few remains of new/old libs on my arduino folder. I just did a clean install of everything (esp32 dev, arduino boards, etc...) and the behaviour changed a bit. Not crashing like before, but still getting a somewhat erratic behaviour, like disconnecting from MQTT. I tried to follow the documentation on the wiki but found that 1) the "complete" guide (https://github.com/espressif/arduino-esp32/blob/master/docs/arduino-ide/debian_ubuntu.md) conflicts with the "short" one (https://github.com/1technophile/OpenMQTTGateway/wiki/Module-Overview-ESP32), as one ask you to get latest version from git and the other a specific version; 2) it seems that my arduino IDE didnt liked the "arduino-esp32" folder, renaming it to esp32 (as the original one) seems to solve it. Right now, I'm using the env you get by following the "short" instructions.  I will update the issue as I test it.

animavitis, 2018-11-06:
Same issue appears for me after updating to https://github.com/1technophile/OpenMQTTGateway/commit/1cb4011a873b52e8796da4544787ec433419e849

1technophile, 2018-11-06:
@animavitis, you get a different behaviour compared to which version ?

animavitis, 2018-11-07:
@1technophile https://github.com/1technophile/OpenMQTTGateway/commit/f77df7347fe8d3e1af2b0551c91586fd8f621bc7 is working fine for me.

I was checking https://github.com/1technophile/OpenMQTTGateway/commit/1cb4011a873b52e8796da4544787ec433419e849 with diferent configuration & always gets GURU MEDITATION at the end:
- BT, RF, RF2, BME280
- BT, RF, BME280
- BT, BME280
- RF, BME280
- BT, RF
- multicore ON
- multicore OFF
- simple publish
- json publish
- simple & json publish

Now im back to 0.8 and all  is working fine

btw.  ReceivedSignal value is always 0 (timestamp is OK) & avoid duplicate dont works when json publish is selected

animavitis, 2018-11-07:
After some more testing i think problem is too big json file, ie. 

`{"id":"58:80:3c::xx:xx:xx","name":"Amazfit Watch-XXXX","rssi":-71,"distance":3.819221,"servicedata":"3022d80100c43b1b3c80d80d","servicedatauuid":"0000fe95-0000-1000-8000-00805f9b34fb"}
Processing BLE device data
E (24433) boot: Assert failed in heap_caps_free, /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/heap/heap_caps.c:268 (heap != NULL && "free() target pointer is outside heap areas")
E (29431) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (29431) task_wdt:  - IDLE0 (CPU 0)
E `

1technophile, 2018-11-07:
Thanks for this analysis. I think you need ti increase : 
#define JSON_MSG_BUFFER 256

To at least 300

animavitis, 2018-11-08:
@1technophile have same error for #define JSON_MSG_BUFFER 512 or #define JSON_MSG_BUFFER 1024

1technophile, 2018-11-08:
and with only BT activated the last master revision is working on your side?

animavitis, 2018-11-08:
after disabling this IF from ZgatewayBT is working stable for 2 hours now..... :
`//            if (advertisedDevice.haveServiceData()){
//              
//                char mac[mac_adress.length()+1];
//                mac_adress.toCharArray(mac,mac_adress.length()+1);
//                
//                trc(F("Get service data "));
//                std::string serviceData = advertisedDevice.getServiceData();
//                int serviceDataLength = serviceData.length();
//                String returnedString = "";
//                for (int i=0; i<serviceDataLength; i++)
//                {
//                  int a = serviceData[i];
//                  if (a < 16) {
//                    returnedString = returnedString + "0";
//                  } 
//                  returnedString = returnedString + String(a,HEX);  
//                }
//                
//                char service_data[returnedString.length()+1];
//                returnedString.toCharArray(service_data,returnedString.length()+1);
//                service_data[returnedString.length()] = '\0';
//                BLEdata.set("servicedata", service_data);
//                
//                BLEdata.set("servicedatauuid", (char *)advertisedDevice.getServiceDataUUID().toString().c_str());
//                
//                char topic[] = subjectBTtoMQTT;
//                strcat(topic, mac);
//                pub(topic,BLEdata);
//                
//                if (strstr(BLEdata["servicedatauuid"].as<char*>(),"fe95") != NULL){
//                  trc("Processing BLE device data");
//                  int pos = -1;
//                  pos = strpos(service_data,"209800");
//                  if (pos != -1){
//                    trc("mi flora data reading");
//                    boolean result = process_data(pos - 24,service_data,mac);
//                  }
//                  pos = -1;
//                  pos = strpos(service_data,"20aa01");
//                  if (pos != -1){
//                    trc("mi jia data reading");
//                    boolean result = process_data(pos - 26,service_data,mac);
//                  }
//                }
//            }`

it looks like problem is somewhere in this part

animavitis, 2018-11-08:
> and with only BT activated the last master revision is working on your side?

same error wth olny BT enabled in config

1technophile, 2018-11-08:
Looks like a good track, thanks. The next step is to reduce the scope of the commented code to isolate the guilty function.

animavitis, 2018-11-08:
I switch my smartwatch to airplane mode to check if OMG will work and OMG is not crashing when my watch is not sending BLE services..... 

For sure OMG failing on parsing BLE services

animavitis, 2018-11-09:
have solution:)  will create PR later :)

1technophile, 2018-11-10:
@clyra could you try the last master revision and see if you have still a crash?

clyra, 2018-11-10:
sure!
 Thanks!

clyra, 2018-11-19:
Hi,

Running for 5 days. It seems to be working right!

1technophile, 2018-11-19:
thanks for the feedback, I close the issue, if you see something new don't hesitate to give feedback

closed 2018-11-19 by 1technophile

maxi1134, 2018-12-18:
Clyra, your commits do not work at all for me. Keep getting a crash every 10 seconds. Anything else you changed?

1technophile, 2018-12-18:
Please open a different issue with more precisions about your environment/versions