1technophile/OpenMQTTGateway #243: Cannot get payloads from rf device???

jascdk, 2018-07-29:
Hi. Just flashed a sonoff bridge with the openmqttt firmware. It flashes perfectly - connects to my wifi and mqtt. I just cant get some payloads out of it. If i subscribe to the topic - pointed out in dr zzs video, it does not get any payload messages, when i send an rf signal from an remote to the bridge???

1technophile, 2018-07-29:
Hi,

Did you tested this remote before flashing with sonoff rf bridge and was it working ? 
Do you see the led on of the rf bridge when you press the remote control ?  And finaly did you change the littke dip switch position once flashed ?

jascdk, 2018-07-29:
Yes - I first tried the tasmota firmware and that worked- it is just not suited for home assistant. Yes the switch is on and the red light on the board lights up when I press a button on the remote

jascdk, 2018-07-29:
I also got the problem with seeing the messages when connection of the seriel monitor - nothing shows up?:(

1technophile, 2018-07-29:
Are ypu sure your serial monitor is at the correct bauds rate  ?  Should be 19200

jascdk, 2018-07-29:
Yes :)

1technophile, 2018-07-30:
Did you changed the switch position ?

1technophile, 2018-07-30:
Could you post what you re seeing in your mqtt broker related to OMG ?

jascdk, 2018-07-30:
Right now I am fiddling with the tasiota firmware - also causes problems - I get back to you when I have reflashed:) Ok?

1technophile, 2018-09-07:
I'm closing this issue , feel free to reopen it if you need

closed 2018-09-07 by 1technophile

Donderda, 2018-10-12:
Hi, I've got the same problem. Flashed Sonoff RF with OpenMQTT, the red LED flashes when I press a button on my remote (I use the official Sonoff RF [Remote](https://www.gearbest.com/smart-home-controls/pp_1824906.html).

But my Serial Monitor works fine :). I also was able to add an IR receiver which works pretty well. The commands are send to my MQTT broker. Do you have any hints what I can try?

Donderda, 2018-10-12:
(and yes, the dip switch is switched to "on" ;) )

gregoriohc, 2020-10-22:
Same problem here :(
Did you solved it @Donderda ?
Thanks!