1technophile/OpenMQTTGateway #244: Bin files

adas4190, 2018-07-31:
Hi. It would be awesome if someone compile firmware for srfb for me as I'm having hard time with arduino. I can flash bin files. It possible please? 

Thank You.

adas4190, 2018-07-31:
Thank you so much!!

adas4190, 2018-07-31:
That link is it to the latest revision? I might have a faulty hardware as it sometimes refuse to connect to wifi or do not go to ap mode. It would be lovely if we could restore factory settings by holding the button for 40sec like in tasmota. 

I love the firmware nevertheless is ultra snappy. Thank You.

1technophile, 2018-07-31:
Could you try[ this one](https://drive.google.com/file/d/1Tl5mViteJ4cqhSbBt0HVls0j6k23kiia/view?usp=sharing), it is the last revision:

1technophile, 2018-07-31:
it is in the [wiki](https://github.com/1technophile/OpenMQTTGateway/wiki/User-guide-SRFB#send-data-by-mqtt-to-convert-it-on-rf-signal)

adas4190, 2018-07-31:
I think I've tried everything but no matter what topic I have used I cannot turn on my cheap rf sockets:(

1technophile, 2018-08-01:
When you press the button of your remote do you see something on your broker ?

adas4190, 2018-08-01:
I can see and use devices but I can only receive codes but I'm stugling to send the code. I tried to serial connection but I'm neeb and I couldny find proper explanation how to do so.

1technophile, 2018-08-03:
So as to send code you need to publish data to your mqtt broker ( not to your serial connection), did you tested the commands pointed in the wiki above ?

adas4190, 2018-08-03:
Ok. I've solved it. It turns out that device (cheat rf outlets) doesnt react when I send the code. I will have to tinker with advenced code options probably. Thank You

1technophile, 2018-08-05:
If ok for you with the bin files provided could you close the issue.

closed 2018-08-06 by adas4190