1technophile/OpenMQTTGateway #246: Auto WiFi dropout reboot

stripeyhorse, 2018-07-31:
**Is your feature request related to a problem? Please describe.**
Can we make it so that if wifi fails a few times (or cant see MQTT) it reboots..

**Describe the solution you'd like**
ping host and if it can 5 times reboot.

**Describe alternatives you've considered**
none yet..

**Additional context**
My WIFI is good. unifi gear. but i keep playing around with it.. if the units could reboot after 5 missed pings it would be great.. (config option for ping host and amount of pings would be great)

1technophile, 2018-07-31:
Did you tested V0.8beta, the wifi connection handling has been improved.

stripeyhorse, 2018-08-01:
silly question. where could I find the V0.8 beta @1technophile

1technophile, 2018-08-03:
On the code page, just download the last revision.

1technophile, 2018-08-22:
@stripeyhorse 
Did you solved the issue by taking the last version?

skeptizzle, 2018-08-28:
On 0.8 beta, i tried 2 different ESP32, both drops connect after some hours. I have Unifi gear as well, could be great with an auto reconnect.
I only use the BT gateway by the way.

1technophile, 2018-08-28:
@skeptizzle 
what version of esp32 env are you using?

skeptizzle, 2018-08-28:
@1technophile
I'll get the version tomorrow, my arduino ide setup is on my workstation at work, but I followed this post (by you). 
https://community.openmqttgateway.com/t/compile-error-on-esp32-on-version-0-7-and-0-8/40/6

skeptizzle, 2018-08-29:
@1technophile 
The ESP32 env is as follows: v3.2-dev-39-gaaf12390

stripeyhorse, 2018-08-30:
@1technophile  sorry. i have since moved my sensors onto Raspberry Pi Ws. Hope to revisit this in the future thou. Thanks Again.

1technophile, 2018-09-07:
@skeptizzle couyld you try with the working configuration identified below:
https://community.openmqttgateway.com/t/ble-and-esp32-dev-environment/102/2

skeptizzle, 2018-09-08:
@1technophile i made a Pi Zero W running, works flawlessly. I will try again in the future.

1technophile, 2018-09-09:
Ok, i will close the issue, if ko for you don't hesitate ti reopen it

closed 2018-09-09 by 1technophile