1technophile/OpenMQTTGateway #251: mijia temperature and humidity sensor | not working

aHVzY2g, 2018-08-18:
**Describe the bug**
The ESP32 isn't able to connect or gather data from MJ_HT_V1. The Serial Monitor reports:
`BTtoMQTT OK
E (336978) BT: btc_search_callback  BLE observe complete. Num Resp 0
`
I experince this issue with 3 new MJ_HT_V1. Only sometimes the ESP32 is able to pick up the name but isn't transmitting any environment data.

**To Reproduce**
Steps to reproduce the behavior:
1. Set up a Windows 10 VM with Guest Tool installed
2. Follow the manuel https://github.com/1technophile/OpenMQTTGateway/wiki/Module-Overview-ESP32
3. Apply Troubleshooting guide Regarding ESP32 arduino environment https://github.com/1technophile/OpenMQTTGateway/wiki/Troubleshooting#compilation-error
4. Edit user_config for wifi details
5. Flash to ESP32 and open Serialport
6. No Results or only the sensor name shows up

**Expected behavior**
ESP32 Reports the Humidty or Temperature from the sensor and prints it to serial or mqtt

**Environment (please complete the following information):**
* OpenMQTTGateway version used (V0.7)
* Windows 10

1technophile, 2018-08-19:
Hello,

Mi jia reading is not possible with v0.7 , it has been implemented on the master branch, (v0.8beta) and not still released. 
If you want to test it you can download it from the [master branch](https://github.com/1technophile/OpenMQTTGateway/archive/master.zip).

aHVzY2g, 2018-08-19:
Thank you for the reply,

I tried now version v0.8beta, I think I got a bit further but still no temprature value. The ESP32 reports on serial can't read data_length or can't read values.

This is the serial Print out:
Connecting to 
not4all@down
.
.
WiFi ok with manual config credentials
OpenMQTTGateway mac: 
30:AE:A4:4E:94:C8
OpenMQTTGateway ip: 
3389040832
1883
Connecting to MQTT by IP adress
192.168.0.38
ZgatewayBT singlecore ESP32 setup done 
MQTT connection...
Connected to broker
Subscription OK to the subjects
Get ManufacturerData 

Get RSSI 
home/OpenMQTTGateway/BTtoMQTT/88C62681A9A2 -66
Get Name 
MJ_HT_V1
Get RSSI 
home/OpenMQTTGateway/BTtoMQTT/4C65A8D3E793 -65
Get service data 
5020aa01c493e7d3a8654c0d100425019201
Get service data UUID
0000fe95-0000-1000-8000-00805f9b34fb
Processing mi flora data
can't read data_length
BTtoMQTT OK
Get ManufacturerData 

Get RSSI 
home/OpenMQTTGateway/BTtoMQTT/88C62681A9A2 -63
Get ManufacturerData 

Get RSSI 
home/OpenMQTTGateway/BTtoMQTT/88C6267AC63A -89
Get RSSI 
home/OpenMQTTGateway/BTtoMQTT/4C65A8D3E793 -56
Get service data 
5020aa01c693e7d3a8654c0d100426019001
Get service data UUID
0000fe95-0000-1000-8000-00805f9b34fb
Processing mi flora data
can't read data_length
BTtoMQTT OK
Remaining memory
30052
Get ManufacturerData 

Get RSSI 
home/OpenMQTTGateway/BTtoMQTT/88C62681A9A2 -62
Get ManufacturerData 
L
Get RSSI 
home/OpenMQTTGateway/BTtoMQTT/68330AB6DB4E -64
BTtoMQTT OK
Get ManufacturerData 

Get RSSI 
home/OpenMQTTGateway/BTtoMQTT/88C62681A9A2 -66
Get ManufacturerData 
L
Get RSSI 
home/OpenMQTTGateway/BTtoMQTT/68330AB6DB4E -63
Get ManufacturerData 

Get RSSI 
home/OpenMQTTGateway/BTtoMQTT/88C6267AC63A -89
Get RSSI 
home/OpenMQTTGateway/BTtoMQTT/4C65A8D3E793 -58
Get service data 
5020aa01d293e7d3a8654c0410022601
Get service data UUID
0000fe95-0000-1000-8000-00805f9b34fb
Processing mi flora data
can't read data_length
BTtoMQTT OK
Get ManufacturerData 

Get RSSI 
home/OpenMQTTGateway/BTtoMQTT/88C62681A9A2 -62
Get ManufacturerData 
L
Get RSSI 
home/OpenMQTTGateway/BTtoMQTT/68330AB6DB4E -65
Get RSSI 
home/OpenMQTTGateway/BTtoMQTT/4C65A8D3E793 -56
Get service data 
5020aa01d793e7d3a8654c0610028201
Get service data UUID
0000fe95-0000-1000-8000-00805f9b34fb
Processing mi flora data
5
0.00
can't read values

(For some reason "Insert code" didn't work)

aHVzY2g, 2018-08-19:
All right,

I recompiled and uploaded version 0.8beta now, and I get the tem and hum values. But I don't get them really regular as you see in the serial console log attached. There seems to be an isue in the decoding process, as I can see sometimes values that are maybe correct but have the comma at the wrong postion (e.g. line 14).

Serial print out:
https://haste.tchncs.de/ohoxuluhag.vbs

1technophile, 2018-08-20:
Here is a comment from another user:
Hi @1technophile
I played around with a Xiaomi Mi Jia Temp & Hum BT sensor on ESP32 OMG and I think that the values are inconsistently decoded (I don't have yet a Flora to be able to compare)
After reviewing the values from serial I think that they are not decoded in entirety. Length for some is 32 (decoded as either hum or temp and they're correct compared to the device), 30 (?) or 36 (these are rejected by OMG - can't read data). Although larger messages (with the length of 36) are rejected and do not appear in MQTT (only RSSI and name are published), I think these are, in fact correct (or nearly correct :) ). Considering that the device is sending, once per day, the battery level and the firmware version, I think that the messages should be at least 40 in length: https://pastebin.com/3THLMPfZ
For the shorter messages that are being decoded, positions 16&15 gives either the temp or hum however my theory is that 16&15 should be always decoded as temp and 18&17 as hum. I'm unsure where battery level goes :)

GijsvanDulmen, 2018-08-28:
I have the same difference in data length and the message "can't read values". I'm, on 0.8beta as well. Will try to capture a log. My Flower Care sensors don't decode as well. Could be related, maybe not.

PetricaM, 2018-09-01:
Hi Florian,

Xiaomi recently pushed a firmware update for both Jia and Flora and https://www.youtube.com/watch?v=0VkrUG3OrPc :)

I'm using 0.8 beta. Actually haven't made any modification to the gateway since initial setup. Moisture and Fertility are the least updated values (I think at about 20-30 minutes) however I'm unsure if this is the intended functionality (to send them only when large variations occurred as it is unlikely these values would rapidly change) or the BLE gateway doesn't decode correctly.

Jia instead is a little more erratic as both hum and temp are being updated on a weird pattern: sometimes after a 1 min interval and some other times after 30 mins or more. RSSI value is updated at 60s sharp. Jia is definitely not sending hum and temp values only when delta is above a certain threshold as I've seen large variations not being pushed to BLE gateway.

1technophile, 2018-09-01:
Hi,

the current principle of decoding value is to use the service data exposed by the sensors,
service data are diffused by the sensor regularly and at sensor discretion.
Each time OMG scan the sensors it retrieves the corresponding service data, one service data sending correspond to one sensor value (temp, hum, moisture...)
The equation between regular OMG retrieving and sensor data exposition can I think lead to inconsistent frequency retrieving of values.

1technophile, 2018-09-13:
From[ the work](https://github.com/1technophile/OpenMQTTGateway/commit/f7eb0b1921a3a27b1f24aa028865c57b6358c99c) of @f-reiling we are now decoding temperature and moisture more frequently (when mi jia send both at the same time) . From my test I'm gathering temp and hum at 1 scan on 2 now. So thanks @f-reiling. Note that battery level has been added for mi jia also.

aHVzY2g, 2018-09-16:
@1technophile Thanks I will do that as soon as possible. At the moment my main PC is "Brocken" and I don't have time to fix it. Could someone else try it?, or someone could send me a precompiled bin and I will try that one.

1technophile, 2018-09-16:
@aHVzY2g with the v0.8 you have a bin file for the esp32 enclosed with the release.
https://github.com/1technophile/OpenMQTTGateway/releases/tag/v0.8

aHVzY2g, 2018-09-17:
All right I can confirm it's decoding now more regular. Thanks for your great work.

Note: I coulnd't find the precompiled version for the ESP32 but I had time to set up a VM and recompile it.

closed 2018-09-17 by aHVzY2g