1technophile/OpenMQTTGateway #260: Home assistant mqtt_room support

umairyounus, 2018-08-29:
Is it possible to use mqtt_room for BLE presence?

https://github.com/home-assistant/home-assistant/blob/master/homeassistant/components/sensor/mqtt_room.py

1technophile, 2018-08-30:
I don't think someone has published the integration yet with mqtt_room

jptrsn, 2018-10-16:
@1technophile I've managed to get some success in bringing functionality into the ESP32 platform. You can take a look at [the repository](https://github.com/jptrsn/ESP32-mqtt-room) and let me know if you're interested. If so, I'll see if I can put together a coherent pull request.

1technophile, 2018-10-16:
Hi @jptrsn , great proposition!
I'm interested (and not only me I think ;-)) If you would like to submit a PR please do it from [refactoring of publishing branch](https://github.com/1technophile/OpenMQTTGateway/tree/publish-refactoring)
I will give you a go when I think the refactoring is enough advanced to make your own branch.
Moreover I'm interested to see if you have problem with watchdog timer, we can discuss about that [here ](https://community.openmqttgateway.com/t/ble-and-esp32-dev-environment/102/52)

jptrsn, 2018-10-16:
Let me know and I'll create a branch off the refactoring of publishing branch and make some changes. I've had the watchdog issue, and I'm currently testing a fix using an async mqtt library.

1technophile, 2018-10-21:
@jptrsn I have made good progress in publishing refactoring regarding the XXtoMQTT direction, you can make a branch from [it](https://github.com/1technophile/OpenMQTTGateway/tree/publish-refactoring) 

As you maybe pointed already all the BLE functions are into [ZgatewayBT.ino](https://github.com/1technophile/OpenMQTTGateway/blob/publish-refactoring/ZgatewayBT.ino)

I will continue the development regarding MQTTtoXX but I don't think it will  affect mqtt_room support. development.

animavitis, 2018-10-29:
Hi

This can be done by adding to **config_BT.h:**

`#define subjectHomePresence "/homePresence"` 

and adding additional code to **ZgatewayBT.ino**:

            if (advertisedDevice.haveRSSI()){
              trc(F("Get RSSI "));       
              String rssi = String(advertisedDevice.getRSSI());
              String rssitopic = mactopic + subjectBTtoMQTTrssi;
              trc(rssitopic + " " + rssi);
              client.publish((char *)rssitopic.c_str(),(char *)rssi.c_str());

      StaticJsonBuffer<200> jsonBuffer;
      JsonObject& HomePresence = jsonBuffer.createObject();
      trc("BLE id :");    
      String id = mac_adress;
      trc(mactopic);
      HomePresence["id"] = id;
      trc("BLE Name :");
      String name;
       if (advertisedDevice.haveName()){
        String name = advertisedDevice.getName().c_str();
       } else {
      String name = mactopic;
      }
      HomePresence["name"] = name;
      trc(name);
      trc("BLE DISTANCE :");
      double ratio = advertisedDevice.getRSSI()/-59;
      double distancetemp = (0.89)* pow(ratio,7.7095) + 0.11;  
      char distance[sizeof(distancetemp)+1];
      dtostrf(distancetemp,2,3,distance);
      HomePresence["distance"] = distance;
      trc(distance);
      char JSONmessageBuffer[100];
      HomePresence.printTo(JSONmessageBuffer, sizeof(JSONmessageBuffer));
      client.publish(subjectHomePresence,JSONmessageBuffer);
            }

This was tested by me and works:)

1technophile, 2018-11-09:
thanks @animavitis

closed 2018-11-09 by 1technophile