1technophile/OpenMQTTGateway #261: SRX882 not working with NODEMCU

umairyounus, 2018-08-29:
I'm trying to connect SRX882 with nodeMCU v1.3 and it's not receiving any code. It's working fine with FS1000A and XD RF. Any idea?

joe63, 2018-08-30:
Ich think You have time Connect also Pin 4(cs) from The srx882.

http://www.rflink.nl/blog2/wiring

joe63, 2018-08-30:
I used fs1000a Wirth arduino IT worked. Then i changed IT with srx882. Didnt Work until connecting Pin 4 .

umairyounus, 2018-08-30:
Hi Joe,

Thanks for a prompt reply. Where did you connect the pin 4? I have tried both ground and 5v.

Thanks

Umair

1technophile, 2018-08-31:
cHi could you post photo if your setup to the forum as this kind of setup has been proven to work I prefer to discuss it on the [community](https://community.openmqttgateway.com)

closed 2018-08-31 by 1technophile

MaartenV87, 2019-03-11:
How can I do this as described on the information page ( https://github.com/1technophile/OpenMQTTGateway/wiki/ESP8266-RF-Send-and-Receive ):

> With SRX882 some users reported that D3 is not working use D1 instead in this case and modify config_RF.h accordingly. 

To what do I have to change the value? in that config?
#ifdef ESP8266
    #define RF_RECEIVER_PIN 0 // D3 on nodemcu
    #define RF_EMITTER_PIN 3 // RX on nodemcu if it doesn't work with 3, try with 4 (D2)

1technophile, 2019-03-11:
take a look to this it gaves you the different possibilities, added to the wiki you should find the necessary infos:
https://docs.google.com/spreadsheets/d/1_5fQjAixzRtepkykmL-3uN3G5bLfQ0zMajM9OBZ1bx0/edit#gid=1617051124

HanSyt, 2020-01-09:
Have spend some time on this, I think that there are a view things to be done to get it working.
Conditions here:
Wemos D1 MIni - SRX882

change config_RF.h to D1 (5)
change the layout of the sketch: connect pin D1 to pin 5 of the srx882 (that's the data pin)
connect pin 4 of the SRX882 (CS) to +5 Volt

In this way I got data

[image]

1technophile, 2020-03-09:
When you say old versions, you mean old versions of your 433mhz devices?

If you want to recognize KAKU for the moment you need to use ZgatewayRF2 instead of ZgatewayRF.

Regarding the Digoo sensors, could you indicate the model of yours?