1technophile/OpenMQTTGateway #262: Clean flashing write-up

thundergreen, 2018-08-30:
First of all thanks for this awesome project.

But I'm stuck at installing / flashing this project to my esp32. I follow the instructions but then it starts:

1. where to copy the libraries in? (I downloaded Arduino and start it after using the install.sh script.)
2. I get always this error: 
```
/home/thorsten/Arduino/libraries/ESP32_BLE_Arduino/src/BLEServer.cpp:21:22: fatal error: gatt_api.h: No such file or directory
compilation terminated.
exit status 1
Fehler beim Kompilieren für das Board ESP32 Dev Module.
```

COuld we plese get a good step by step instruction how to proceedfrom zero for users without any knowledge?

Thanks in advance

P.S I got it already flashed somehow but broke something with another project. But now after reinstalling everything new.. I am really stuck

HELP needed :)

1technophile, 2018-08-31:
1. the content of the lib folder from the repo should be copied to the libraries folder of your arduino ide (the sketchbook folder on which the libraries folder is located, is defined in your IDE preferences)
2. this error is due to the fact that you are not using the good version of BLE library or esp32 arduino

Please note that I'm currently donwloading the last version of arduino esp32 stable to try it and maybe redo the instructions if it works

1technophile, 2018-09-02:
Hello,

Here is the working combination of esp32 env and library (repo and wiki updated)
[From this commit](https://github.com/espressif/arduino-esp32/tree/89859f7f4c07ce3e5d487a4b689dbda78f5d5c3a)  with this[BLE library version](https://github.com/nkolban/ESP32_BLE_Arduino/tree/7951347ed68313d75c367e1f2cce763cb56d1eb2) 

On the arduino ide choose Minimal SPIFFS as partition scheme

thundergreen, 2018-09-02:
The PRoblem is that the order of the installation won'Twork.. If I follow your guide Arduino IDE tells m that no hardware information has been found in /Arduino/hardware/espressif/....

Normal As the instal instructions tell me to download arduino-esp in hardware folder.. so it's a bit confusing what to do when.

COuld you please just make a clean write up with step by step for stupid dumbs like me ?

Sorry to bother you for this silliness

thundergreen, 2018-09-02:
how do i have to install this esp-arduino thing.. it's annoying.. installation creates each time libraries folders and arduino ide searches everywhere and i duno where t put what.. Could you really please make a dumb step by step install discription.. I'm freaking out here

1technophile, 2018-09-02:
1. Uninstall your previous arduino esp32 version (remove folders)
1. Download the [Arduino ESP32 environment](https://github.com/espressif/arduino-esp32/archive/89859f7f4c07ce3e5d487a4b689dbda78f5d5c3a.zip) note that newer versions of esp32 env [doesn't work](https://github.com/nkolban/esp32-snippets/issues/592), you must take the ones linked, install it following the Instructions below :
1. Locate your sketchbook folder; open your arduino IDE, go to file->preferences, copy the path of your sketchbook folder

[image]

1. Extract the folder into <sketchbook folder>/hardware/espressif
1. Rename the folder extracted into espressif folder arduino-esp32
1. Then go to <sketchbook folder>/hardware/arduino-esp32/tools
and run: sudo python get.py(linux) or get.exe (windows)
1. Download the OpenMQTTGateway code with all the files and put them on a folder named `OpenMQTTGateway`. Open the file `OpenMQTTGateway.ino`
1. Choose board ESP32 dev module
1. Change partition scheme to minimal SPIFFS (if you don't do that you will not have enough space for your program), if your board is not caompatible with ESP32 dev module choice you can modify [parameters manualy](https://github.com/1technophile/OpenMQTTGateway/wiki/Sketch-too-big,-instructions-to-avoid-this-error-with-BLE-library-(if-not-using-minimal-SPIFFS))
1. Set your parameters into user_config.h
1. Upload the code

thundergreen, 2018-09-02:
Wow...thanks for dummy write up. You are the best ..it's compiling now 😎

1technophile, 2018-09-03:
Hmm nope, you mean the name of the sensor?

thundergreen, 2018-09-11:
Ignore my last comment. Everything working fine ..issue solved ... User happy... Software runs ...--> ✓ closed

closed 2018-09-11 by thundergreen

pauldogg, 2018-11-23:
I too am struggling with this for quite some time.
How did you get Arduino to see an Esp32 dev module?
The instructions are a bit confusing for me... I get lost at steps 4 and 5

Is there a Bin file that we can flash using esp32 flasher?
Thanks in advance!

1technophile, 2018-11-23:
sketchbook folder is located into your arduino ide preferences,
your env structure should look like that:
[image]