1technophile/OpenMQTTGateway #263: TimeBtw_Read more than 60000

Destix, 2018-09-06:
**Describe the bug**
Then TimeBtw_Read set to more than 60000 i'm geting no readings. Only Remaining memory messages in serial monitor window. Tried to play with Scan_duration setting, does not make any difference.

**To Reproduce**
Set TimeBtw_Read to more than  60000 
Tried 300000 (once in 5 minutes)
Actually i need read settings once 20-30 minutes (the would be 1,800,000 Milliseconds)  to save Mi Flora battery life.

**Enviroment**
OpenMQTTGateway 0.7
ESP32 (LoLin32 bought on AliExpress)
Sensor: Mi Flower
Compiled on arduino IDE 1.8.5

1technophile, 2018-11-11:
@Destix could you test it with the last master code?

1technophile, 2019-06-21:
I'm closing this issue, if you need other info don't hesitate to reply

closed 2019-06-21 by 1technophile