1technophile/OpenMQTTGateway #270: ZactuatorONOFF, issue to connect to MQTT broker

salvq, 2018-09-14:
Issue to connect to MQTT broker when ZactuatorONOFF defines used, rest defines commented out. MQTT publish does not work, omg is trying to connect to broker in loop without sucess
[image]

When disable ZactuatorONOFF and uncomment DHT (rest defines are commented out), no problem to connect to broker and working just fine.
[image]

salvq, 2018-09-14:
I am having Arduino UNO, W5100 shield, if you need any addtional information just let me know

OMG_VERSION 0.8
char mqtt_user[20] = "123"; // not compulsory only if your broker needs authentication
char mqtt_pass[20] = "456"; // not compulsory only if your broker needs authentication
char mqtt_server[40] = "192.168.1.112";
char mqtt_port[6] = "1883";

const byte ip[] = { 192, 168, 1, 114 }; //ip adress
const byte gateway[] = { 192, 168, 1, 59 }; //ip adress, if first value is different from 0 advanced config network will be used and you should fill gateway & dns
const byte Dns[] = { 192, 168, 1, 59 }; //ip adress, if first value is different from 0 advanced config network will be used and you should fill gateway & dns
const byte subnet[] = { 255, 255, 255, 0 }; //ip adress

1technophile, 2018-09-14:
@salvq do you have several gateways called OpenMQTTGateway (or with the same name)

salvq, 2018-09-14:
Just one overall, I have been switching between DHT and ONOFF on the same GW.

1technophile, 2018-09-14:
Could you try others pins like 8, 9 or 2 for ACTUATOR_ONOFF_PIN?

salvq, 2018-09-14:
Just did, same over and over again, strange thing is it will not even finish sentence like in sketch trc(F("ZactuatorONOFF setup done ")); ...just ZactuatorONOFF setup

[image]

[image]

[image]

salvq, 2018-09-14:
Let me also try use MEGA instead of UNO to see if any difference...

salvq, 2018-09-14:
It is related to UNO, with same settings just different board arduino MEGA it is working fine.

- UNO (DHT working OK, ONOFF can not get connected)
- MEGA (DHT working OK, ONOFF working OK)

[image]

1technophile, 2018-09-15:
Could you give your compilation results on UNO (the end of compilation messages) like remaininh memory ...

salvq, 2018-09-15:
Here you go. I also did a test with IP settings and looks like it works if GW and DNS are 0's. Please verify from your end how this is related / could be automated otherwise add comment to IP addresses to leave 0's for UNO as default...

Fixed IP settings (can not get connection up with UNO)
[image]

Gateway and DNS to zero (working ok with UNO)
[image]

closed 2019-08-14 by stale[bot]