1technophile/OpenMQTTGateway #272: Sketch compilation issue, nodemcu

salvq, 2018-09-18:
I am trying to flash opengatewaymqtt v0.8 to nodemcu v3 (esp-12n) and getting following error. What I am doing wrong ? Simple sketch is working fine (serial printing or LED blink).

See my setup:

```
define OMG_VERSION "0.8"

/*-------------DEFINE YOUR MQTT PARAMETERS BELOW----------------*/
//MQTT Parameters definition
//#define mqtt_server_name "www.mqtt_broker.com" // instead of defining the server by its IP you can define it by its name, uncomment this line and set the correct MQTT server host name
char mqtt_user[20] = "XZY"; // not compulsory only if your broker needs authentication
char mqtt_pass[20] = "ZYZ"; // not compulsory only if your broker needs authentication
char mqtt_server[40] = "192.168.10.89";
char mqtt_port[6] = "1883";

#define Gateway_Name "OpenMQTTGateway"
#define Base_Topic "home/"
#define version_Topic  Base_Topic Gateway_Name "/version"
#define will_Topic  Base_Topic Gateway_Name "/LWT"
#define will_QoS 0
#define will_Retain true
#define will_Message "Offline"
#define Gateway_AnnouncementMsg "Online"

/*-------------DEFINE YOUR NETWORK PARAMETERS BELOW----------------*/

#define ESPWifiManualSetup true //uncomment you don't want to use wifimanager for your credential settings on ESP
#define WifiManager_password "password"
//#define MDNS_SD //uncomment if you  want to use mdns for discovering automatically your ip server, please note that MDNS with ESP32 can cause the BLE to not work
//#define cleanFS true //uncomment if you want to clean the ESP memory and reenter your credentials
#define maxMQTTretry 4 //maximum MQTT connection attempts before going to wifi setup

//set minimum quality of signal so it ignores AP's under that quality
#define MinimumWifiSignalQuality 8

// Update these with values suitable for your network.
#if defined(ESP32) || defined(ESPWifiManualSetup) // for nodemcu, weemos and esp8266
  #define wifi_ssid "2G_AWE"
  #define wifi_password "passwordtest"
#else // for arduino + W5100
  const byte mac[] = {  0xCE, 0xBA, 0xBA, 0xFA, 0x53, 0x95 }; //W5100 ethernet shield mac adress
#endif

const byte ip[] = { 192, 168, 1, 150 }; //ip adress
const byte gateway[] = { 192, 168, 1, 59 }; //ip adress, if first value is different from 0 advanced config network will be used and you should fill gateway & dns
const byte Dns[] = { 192, 168, 1, 59 }; //ip adress, if first value is different from 0 advanced config network will be used and you should fill gateway & dns
const byte subnet[] = { 255, 255, 255, 0 }; //ip adress

/*-------------DEFINE YOUR OTA PARAMETERS BELOW----------------*/
#define ota_hostname Gateway_Name
#define ota_password "otapassword"
#define ota_port 8266

/*-------------DEFINE PINs FOR STATUS LEDs----------------*/
#define led_receive 40
#define led_send 42
#define led_error 44
```

Arduino: 1.8.6 (Windows 10), Board: "NodeMCU 1.0 (ESP-12E Module), 80 MHz, Flash, 4M (1M SPIFFS), v2 Lower Memory, Disabled, None, Only Sketch, 115200"

```
Z:\Home_Automation\OpenMQTTGateway\OpenMQTTGateway.ino: In function 'void stateMeasures()':

OpenMQTTGateway:693:7: error: 'StaticJsonBuffer' was not declared in this scope

       StaticJsonBuffer<200> jsonBuffer;

       ^

OpenMQTTGateway:693:29: error: 'jsonBuffer' was not declared in this scope

       StaticJsonBuffer<200> jsonBuffer;

                             ^

OpenMQTTGateway:762:15: error: 'class ArduinoJson::JsonObject' has no member named 'printTo'

       SYSdata.printTo(JSONmessageBuffer, sizeof(JSONmessageBuffer));

               ^

exit status 1
'StaticJsonBuffer' was not declared in this scope

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

```

1technophile, 2018-09-18:
I think you missed to copy the arduinojson library, please refer to the wiki

salvq, 2018-09-18:
I was looking into this before posting and I do have it installed...any other check to be done ? I read the whole wiki and did not find what I am doing wrong.

[image]

1technophile, 2018-09-19:
You qhould not use the arduino library manager but just cipy the libraries from openmqttgateway/lib folder to your sketchbook folder/libraries. You can also download the esp8266 bin file and flash it directly to the chip.

salvq, 2018-09-21:
Thank you sir for guidance, I will be more careful next time. It is working like a charm.

Your work and support on the Gateway is really great...

bphermansson, 2018-09-25:
That feels like to beg for trouble. Or at least library version conflicts?

TT8174, 2018-12-28:
> You should not use the arduino library manager but just copy the libraries from openmqttgateway/lib folder to your sketchbook folder/libraries. You can also download the esp8266 bin file and flash it directly to the chip.

Hi Thanks for your project.
I use Arduino 1.8.8. and I cope openmqttgateway/lib folder to C:\Program Files (x86)\Arduino\libraries,
but I received the same problem.

Thanks.

My wrong info is:
`

C:\Users\litao\Desktop\hass\OpenMQTTGateway-master\OpenMQTTGateway-master\OpenMQTTGateway\OpenMQTTGateway.ino: In function 'boolean reconnect()':

OpenMQTTGateway:201:26: error: 'Gateway_Name' was not declared in this scope

       if (client.connect(Gateway_Name, mqtt_user, mqtt_pass, will_Topic, will_QoS, will_Retain, will_Message)) {

                          ^

OpenMQTTGateway:201:40: error: 'mqtt_user' was not declared in this scope

       if (client.connect(Gateway_Name, mqtt_user, mqtt_pass, will_Topic, will_QoS, will_Retain, will_Message)) {

                                        ^

OpenMQTTGateway:201:51: error: 'mqtt_pass' was not declared in this scope

       if (client.connect(Gateway_Name, mqtt_user, mqtt_pass, will_Topic, will_QoS, will_Retain, will_Message)) {

                                                   ^

OpenMQTTGateway:201:62: error: 'will_Topic' was not declared in this scope

       if (client.connect(Gateway_Name, mqtt_user, mqtt_pass, will_Topic, will_QoS, will_Retain, will_Message)) {

                                                              ^

OpenMQTTGateway:201:74: error: 'will_QoS' was not declared in this scope

       if (client.connect(Gateway_Name, mqtt_user, mqtt_pass, will_Topic, will_QoS, will_Retain, will_Message)) {

                                                                          ^

OpenMQTTGateway:201:84: error: 'will_Retain' was not declared in this scope

       if (client.connect(Gateway_Name, mqtt_user, mqtt_pass, will_Topic, will_QoS, will_Retain, will_Message)) {

                                                                                    ^

OpenMQTTGateway:201:97: error: 'will_Message' was not declared in this scope

       if (client.connect(Gateway_Name, mqtt_user, mqtt_pass, will_Topic, will_QoS, will_Retain, will_Message)) {

                                                                                                 ^

OpenMQTTGateway:205:22: error: 'Gateway_AnnouncementMsg' was not declared in this scope

       pub(will_Topic,Gateway_AnnouncementMsg,will_Retain);

                      ^

OpenMQTTGateway:207:11: error: 'version_Topic' was not declared in this scope

       pub(version_Topic,OMG_VERSION,will_Retain);

           ^

OpenMQTTGateway:207:25: error: 'OMG_VERSION' was not declared in this scope

       pub(version_Topic,OMG_VERSION,will_Retain);
[... 208 lines cut ...]

   StaticJsonBuffer<JSON_MSG_BUFFER> jsonBuffer;

                    ^

OpenMQTTGateway:873:35: error: template argument 1 is invalid

   StaticJsonBuffer<JSON_MSG_BUFFER> jsonBuffer;

                                   ^

OpenMQTTGateway:873:47: error: invalid type in declaration before ';' token

   StaticJsonBuffer<JSON_MSG_BUFFER> jsonBuffer;

                                               ^

OpenMQTTGateway:874:37: error: request for member 'parseObject' in 'jsonBuffer', which is of non-class type 'int'

   JsonObject& jsondata = jsonBuffer.parseObject(datacallback);

                                     ^

exit status 1
'Gateway_Name' was not declared in this scope`

1technophile, 2018-12-28:
did you copied the content of the lib folder into libraries or the lib folder?
You should copy the content of the folder.

TT8174, 2018-12-28:
Thanks frist.

I copy OpenMQTTGateway-master\lib  to Arduino\libraries, So it is ok or not?

1technophile, 2018-12-28:
if you copied the content of lib yes it is ok, if you copied lib it is not ok.
At final you should have for example a pubsubclient folder into Arduino (Arduino\libraries\pubsubclient )and not Arduino\libraries\lib\pubsubclient

TT8174, 2018-12-29:
[image]
yes,I aready copy all files in libraries folder,please see my pictures.
but again I got the same problem.
and I try to search key words(Gateway_Name) in  arduinojson folder,can't find anything.
Thanks.

1technophile, 2018-12-29:
it seems you are missing user_config.h in OpenMQTTGateway.ino folder

TT8174, 2018-12-30:
Problem is solved！
Thanks！Thanks！Thanks！Thanks！
When I frist run OpenMQTTGateway.ino，Arduino IDE will creat a folder named OpenMQTTGateway, and OpenMQTTGateway.ino will be move in this folder. but User_confg.h and other *.h files not. I move this files in the OpenMQTTGateway folder,Every thing is OK,