1technophile/OpenMQTTGateway #273: ITag BLE device button detect

tototek, 2018-09-19:
ITag is cheap BLE device equipped with button and piezzo buzzer.(https://www.amazon.com/Anti-Lost-Bluetooth-Remote-Shutter-Tracker/).
I have tested it and it detects it as beacon correctly and it gives its RSSI . (home / OpenMQTTGateway / BTtoMQTT / FFFF10xxxx ).
It is possible to detect in OpenMQttGAteway, the status of the button, 1 click, 2 clicks, it would be ideal for HA.
I would appreciate any help

AKA-47, 2018-10-29:
I have these but seems like you need an active pairing to get the button status. You can debug wit nRF Connect on Android for example.
In my setup with esp32 the range is really bad so it is almost unusable for presence detection without multiple devices around the House

Misiu, 2019-11-25:
@tototek link you added is not available.
Did the device look like this:
[image]
https://www.aliexpress.com/item/4000096686419.html

I found information that this device turns itself offline when it is not connected to the phone via Bluetooth for 15 minutes - https://thejeshgn.com/2017/06/20/reverse-engineering-itag-bluetooth-low-energy-button/#comment-177416

Is there a way to disable that sleep mode? These devices are very cheap so ideal for many usages.
@1technophile did you had a chance to try such devices in the past?

I'm looking for a way to detect the device when I get back home. If the device turns itself off after 15 minutes it will be useless.

The cheapest device on [the list](https://docs.google.com/spreadsheets/d/1_5fQjAixzRtepkykmL-3uN3G5bLfQ0zMajM9OBZ1bx0/edit#gid=2126158079) is BLE beacon from smart finder (~$4). Does it send messages all the time (won't it turn off after some time)

I didn't want to create a separate issue because I think that I need the same thing as @tototek.

1technophile, 2019-11-26:
> Is there a way to disable that sleep mode? These devices are very cheap so ideal for many usages.
@1technophile did you had a chance to try such devices in the past?

I had this kind of devices and I'm suspicious that indeed they turn off automaticaly. Nevertheless I can't confirm it as both are dead...

> The cheapest device on the list is BLE beacon from smart finder (~$4). Does it send messages all the time (won't it turn off after some time)
I confirm this one doesn't turn off automaticaly.

Misiu, 2019-11-26:
@1technophile thank You so much for the reply. It is really helpful! 
I was thinking about ordering the cheaper ones but now I'm convinced to order those from your list.
Did you do range testing? From how far were they detected by ESP32? Do you remember the results?

1technophile, 2019-11-29:
> Did you do range testing? From how far were they detected by ESP32? Do you remember the results?

Not for the moment but I had it on my todo

Misiu, 2019-12-02:
@1technophile I ordered three different ESP32 modules and those ble beacons.
I'll try them out and if everything goes well I'll share the results.

1technophile, 2019-12-02:
Great!

FedericoHenze, 2019-12-19:
I want to use mine as a key finder, but the problem is, when it goes out of range, it starts beeping, which is annoying. Any ideas to disable this feature? It is on the device itself as a default behaviour, so maybe is not possible to modify it...

1technophile, 2019-12-19:
> I want to use mine as a key finder, but the problem is, when it goes out of range, it starts beeping, which is annoying. Any ideas to disable this feature? It is on the device itself as a default behaviour, so maybe is not possible to modify it...

It's a little bit heavy but maybe remove the buzzer nevertheless I'm afraid that it will quickly run out of battery if it has this behaviour

AKA-47, 2020-10-02:
> It's a little bit heavy but maybe remove the buzzer nevertheless I'm afraid that it will quickly run out of battery if it has this behaviour
@FedericoHenze @1technophile 
**You dont have to remove the buzzer to use the device as presence detector.**

The thing is if you pair the device it starts beeping if you tell it to do so when pairing is lost. This depends on the last command sent to the chip. The default behavior is to start beeping i think.

Use NRFConnect on Android to check which servicedata is exposed by the device because there are different revisions with different firmwares. 
Biggest difference is that some of them have readable battery level.
You can then use it as Keyfinder too and manually send commands like "High Alert" to the device which causes the device to beep then. 

I use 2 different types:
https://www.aliexpress.com/item/4000096686419.html AND https://www.aliexpress.com/item/32908970479.html
The first ones have better range from which i can recieve ble beacon messages.

The presence detections works like this for me in openhab2

items:
`Number ITAGWOHNUNG "Wohnungsschluessel RSSI [%d %%]"  {mqtt="<[broker:home/BLE-GW/BTtoMQTT/FFFFXXXXC393/rssi:state:default]"}
DateTime ITAGWOHNUNG_LastUpdate "ITAGWOHNUNG Zuletzt gesehen [%1$ta %1$tR]" <clock>
`
--> this updates the item ITAGWOHNUNG whis the rssi. ITAGWOHNUNG_LastUpdate gets updateted with timestamp by rule.
A update only happens when beacon is in range.

rules:

```
rule "ITAGWOHNUNG_LastUpdate last seen"
when
  Item ITAGWOHNUNG received update
then
  postUpdate(ITAGWOHNUNG_LastUpdate, new DateTimeType())
end
```

Now you can write a rule which triggers "on update" of this item and for example use the expire binding for prevent flapping to do things or combine the state somehow.

im not sure if this works in OMG 0.9.5 because there were some changes to BLE but before it did.

The thing with the button is that you need an active pairing to get its state. for me the button is useless...

Greets

closed 2023-02-04 by 1technophile