1technophile/OpenMQTTGateway #277: ESP32 Compile Errors

Digitsl-dB, 2018-09-22:
I am trying to compile the code to for the ESP32 board but keep getting errors.
MQTTGateway 0.7, master, 0.8 and lighter version from trouble shooting link.
Arduino 1.8.7.
Downloaded the Arduino ESP32 environment and put it in the sketchbook folder.
Copied libraries from OpenMQTTGateway/lib/ESP32_BLE_Arduino to "sketchbook folder".
I also tried using the previous lighter version as mentioned in the trouble shooting section but get the same error.

Please advise if I am using the wrong library versions or Arduino.... I have no idea what the issue is here.
Thanks

Using library pubsubclient at version 2.6 in folder: C:\Users\chris\Documents\Arduino\libraries\pubsubclient 
Using library ArduinoJson at version 5.13.1 in folder:
C:\Users\chris\Documents\Arduino\libraries\ArduinoJson 
Using library WiFi at version 1.0 in folder: C:\Users\chris\Documents\Arduino\hardware\espressif\arduino-esp32\libraries\WiFi 
Using library ArduinoOTA at version 1.0 in folder:
C:\Users\chris\Documents\Arduino\hardware\espressif\arduino-esp32\libraries\ArduinoOTA 
Using library Update at version 1.0 in folder:
C:\Users\chris\Documents\Arduino\hardware\espressif\arduino-esp32\libraries\Update 
Using library ESP32_BLE_Arduino at version 0.4.8 in folder:
C:\Users\chris\Documents\Arduino\libraries\ESP32_BLE_Arduino 
Using library ESPmDNS at version 1.0 in folder:
C:\Users\chris\Documents\Arduino\hardware\espressif\arduino-esp32\libraries\ESPmDNS 

Arduino: 1.8.7 (Windows 10), Board: "ESP32 Dev Module, QIO, 80MHz, 4MB (32Mb), 115200, None"

<command-line>:0:21: warning: ISO C++11 requires whitespace after the macro name

In file included from C:\Users\chris\Documents\Home\OpenMQTTGateway_V07\OpenMQTTGateway\OpenMQTTGateway.ino:71:0:

sketch\config_BT.h:30:11: warning: ISO C++11 requires whitespace after the macro name

 #define HM-10

           ^

In file included from C:\Users\chris\Documents\Arduino\libraries\ESP32_BLE_Arduino\src/BLEDevice.h:16:0,

                 from C:\Users\chris\Documents\Home\OpenMQTTGateway_V07\OpenMQTTGateway\ZgatewayBT.ino:38:

C:\Users\chris\Documents\Arduino\hardware\espressif\arduino-esp32/tools/sdk/include/bt/bt.h:2:2: warning: #warning "This header is deprecated, please use functions defined in esp_bt.h instead." [-Wcpp]

 #warning "This header is deprecated, please use functions defined in esp_bt.h instead."

  ^

C:\Users\chris\Documents\Home\OpenMQTTGateway_V07\OpenMQTTGateway\OpenMQTTGateway.ino: In function 'boolean reconnect()':

OpenMQTTGateway:175:109: error: no matching function for call to 'PubSubClient::connect(const char [17], char [20], void (&)(char*, byte*, unsigned int), const char [25], int, bool, const char [8])'

       if (client.connect(Gateway_Name, mqtt_user, mqtt_pass, will_Topic, will_QoS, will_Retain, will_Message)) {

                                                                                                             ^

In file included from C:\Users\chris\Documents\Home\OpenMQTTGateway_V07\OpenMQTTGateway\OpenMQTTGateway.ino:60:0:

C:\Users\chris\Documents\Arduino\libraries\pubsubclient\src/PubSubClient.h:125:12: note: candidate: boolean PubSubClient::connect(const char*)

    boolean connect(const char* id);

            ^

C:\Users\chris\Documents\Arduino\libraries\pubsubclient\src/PubSubClient.h:125:12: note:   candidate expects 1 argument, 7 provided

C:\Users\chris\Documents\Arduino\libraries\pubsubclient\src/PubSubClient.h:126:12: note: candidate: boolean PubSubClient::connect(const char*, const char*, const char*)

    boolean connect(const char* id, const char* user, const char* pass);

            ^

C:\Users\chris\Documents\Arduino\libraries\pubsubclient\src/PubSubClient.h:126:12: note:   candidate expects 3 arguments, 7 provided

C:\Users\chris\Documents\Arduino\libraries\pubsubclient\src/PubSubClient.h:127:12: note: candidate: boolean PubSubClient::connect(const char*, const char*, uint8_t, boolean, const char*)

    boolean connect(const char* id, const char* willTopic, uint8_t willQos, boolean willRetain, const char* willMessage);

            ^

C:\Users\chris\Documents\Arduino\libraries\pubsubclient\src/PubSubClient.h:127:12: note:   candidate expects 5 arguments, 7 provided

C:\Users\chris\Documents\Arduino\libraries\pubsubclient\src/PubSubClient.h:128:12: note: candidate: boolean PubSubClient::connect(const char*, const char*, const char*, const char*, uint8_t, boolean, const char*)

    boolean connect(const char* id, const char* user, const char* pass, const char* willTopic, uint8_t willQos, boolean willRetain, const char* willMessage);

            ^

C:\Users\chris\Documents\Arduino\libraries\pubsubclient\src/PubSubClient.h:128:12: note:   no known conversion for argument 3 from 'void(char*, byte*, unsigned int) {aka void(char*, unsigned char*, unsigned int)}' to 'const char*'

Multiple libraries were found for "BLEDevice.h"
 Used: C:\Users\chris\Documents\Arduino\libraries\ESP32_BLE_Arduino
 Not used: C:\Users\chris\Documents\Arduino\hardware\espressif\arduino-esp32\libraries\BLE
exit status 1
no matching function for call to 'PubSubClient::connect(const char [17], char [20], void (&)(char*, byte*, unsigned int), const char [25], int, bool, const char [8])'

1technophile, 2018-09-22:
First step is to follow the wiki for esp32, did you followed it ? 
https://github.com/1technophile/OpenMQTTGateway/wiki/Module-Overview-ESP32

Digitsl-dB, 2018-09-22:
That is the guide I used.

I also tried MQTTGateway 0.7, then master version from GitHub, then 0.8 (latest from GitHub) and the lighter version from trouble shooting link.
I am using Arduino 1.8.7.
I downloaded the Arduino ESP32 environment and put it in the sketchbook folder as shown in the wiki.
Copied libraries from OpenMQTTGateway/lib/ESP32_BLE_Arduino to "sketchbook folder". The skect folder is "C:\Users\chris\Documents\Arduino"

The final error was in OpenMQTTGateway.ino

> no matching function for call to 'PubSubClient::connect(const char [17], char [20], void (&)(char*, byte*, unsigned int), const char [25], int, bool, const char [8])'

I tried different versions of the PubSubClient but have the same error, the full list of libraries is:

pubsubclient at version 2.6
ArduinoJson at version 5.13.1
WiFi at version 1.0
ArduinoOTA at version 1.0
Update at version 1.0
ESP32_BLE_Arduino at version 0.4.8
ESPmDNS at version 1.0

These are the ones included with the GitHub download, are these correct for the ESP32 compile?

1technophile, 2018-09-22:
Yes they are the correct ones, and the ones used by the automatic compilation if you want to compare
https://travis-ci.org/1technophile/OpenMQTTGateway/jobs/431834037

Digitsl-dB, 2018-09-23:
I think the issue was with the mqtt_pass, I commented this out as I don't have a MQTT password but I should have deleted the text "your_password". I started fresh  and downloaded everything again and it compiled. 
Thanks for the support

closed 2018-09-23 by Digitsl-dB