1technophile/OpenMQTTGateway #279: Getting these Error

MYLE-01, 2018-10-02:
```
Arduino: 1.8.7 (Windows 7), Board: "WeMos D1 R2 & mini, 80 MHz, 4M (1M SPIFFS), v2 Lower Memory, Disabled, None, Only Sketch, 115200"

C:\Users\Debbie & StePhan\Documents\Arduino\OpenMQTTGateway\ZgatewayIR.ino: In function 'void MQTTtoIR(char*, char*)':

ZgatewayIR:398:37: error: 'GICABLE_BITS' was not declared in this scope

     if (valueBITS == 0) valueBITS = GICABLE_BITS;

                                     ^

ZgatewayIR:399:65: error: 'GICABLE_MIN_REPEAT' was not declared in this scope

     if (valueRPT == 0) valueRPT = std::max(valueRPT, (uint16_t) GICABLE_MIN_REPEAT);

                                                                 ^

ZgatewayIR:400:12: error: 'class IRsend' has no member named 'sendGICable'

     irsend.sendGICable(data, valueBITS, valueRPT);

            ^

ZgatewayIR:459:12: error: 'class IRsend' has no member named 'sendMitsubishi2'

     irsend.sendMitsubishi2(data, valueBITS, valueRPT);

            ^

exit status 1
'GICABLE_BITS' was not declared in this scope

```

MYLE-01, 2018-10-02:
went back to 0.7 and it working

so must be a problem with 0.8

1technophile, 2018-10-03:
hmm, what is the version of irremoteesp8266 library used?
You should see the version at the end of your arduino IDE compilation messages

1technophile, 2018-10-13:
I m closing this issue, if you still have the problem you can reopen it

closed 2018-10-13 by 1technophile

rockofclay, 2018-10-26:
I was having this issue until I rolled back to irremoteesp8266 2.4.3.

Pityu454, 2018-11-04:
Hello! I got this problem too with irremoteesp8266 v2.5.2, I rolled back to the v2.4.3 too, and it is working again! Thanks rockofclay!

1technophile, 2018-11-04:
My advice is to use the library included into the lib folder. Currently it is the 2.4.2

Pityu454, 2018-11-04:
> My advice is to use the library included into the lib folder. Currently it is the 2.4.2

O, I didn't see it, sorry! And thanks this project, it is fantastic! :)

1technophile, 2018-11-04:
No problem, have fun with it