1technophile/OpenMQTTGateway #280: ESP32 with DHT11 not showing temperature or humidity

Digitsl-dB, 2018-10-04:
Hi I have this working well with the ESP32 for BT data but it is not working for temp or humidity with the DHT11.
I have uncommented the code in the user_config.h and config_DHT.h for the DHT11 sensor (other sensors commented out).
I also changed the gateway name to the location of the device in config_DHT.h like this
`#define HUM1    Base_Topic Gateway_Name "/DHTtoMQTT/loft/hum"`
`#define TEMP1   Base_Topic Gateway_Name "/DHTtoMQTT/loft/temp"`

The DHT_RECEIVER pin I assume is 35 for the ESP32 (as per the code) but I get no data on the MQTT for this. Data from BT is working correctly. 
Looking at the ESP32 datasheet, pin 35 is GPIO18 on the development board.
[ESP32 WROOM-32](https://i.pinimg.com/originals/c6/57/83/c657835e84aaf91832a770ea0d7d0767.jpg)

When I connect the DHT11 to my ESP8266 (running tasmota) the data comes through MQTT. It seems that I might be connecting to the wrong pin on the ESP32 or the DHT code is not running. 
Once the DHT code is enabled in user_config.h should I get data on the MQTT (/temp 0) or (/hum 0)? I do not get this data either.

1technophile, 2018-10-04:
To check your mqtt messages the best way is to subscribe to all data coming as a first approach:
`mosquitto_sub -t +/# -v`
For the DHT try to connect the data to pin 16 and change config_DHT.h accordingly

Digitsl-dB, 2018-10-06:
Hi, 
Connecting to Pin 16 worked, it is also marked as 25, GPIO16 and RXD on the diagram but it is working well.
Thanks

closed 2018-10-06 by Digitsl-dB