1technophile/OpenMQTTGateway #281: mi jia "can't read values"

terafin, 2018-10-07:
Before submitting a problem please check the troubleshooting section
https://github.com/1technophile/OpenMQTTGateway/wiki/Troubleshooting

**Describe the bug**
* Using ESP8266 from here: https://www.amazon.com/gp/product/B01IK9GEQG/ref=oh_aui_search_detailpage?ie=UTF8&psc=1
* H10 from here: https://www.amazon.com/gp/product/B010SFF8QC/ref=oh_aui_search_detailpage?ie=UTF8&psc=1

Both with the v8 bin, as well as a compiled sketch I'm unable to decode things from the mi jia sensors:

```
Activated modules
RFIRBT
4f4b2b444953413af79fd3a8654c00a61a09844d4a5f48545f563105030f180a180916fffffa8b8feff6bd0d0a
4C65A8D39FF7
4C65A8D39FF7
service_data
5f563105030f180a180916fffffa8b8feff6bd
BTtoMQTT OK
4f4b2b444953413af79fd3a8654c00a619020106151695fe5020aa0128f79fd3a8654c0d1004e900ec010d0a
4C65A8D39FF7
4C65A8D39FF7
service_data
5020aa0128f79fd3a8654c0d1004e900ec01
mi jia data reading
3
0.00
can't read values
BTtoMQTT OK
Uptime (s)
960
Remaining memory
33176
RSSI
-39
SSID
Intarweb IOT
Activated modules
RFIRBT

```

**To Reproduce**
Install via v8 bin (or sketch), hook up, and wait :)

**Environment (please complete the following information):**
v8 from source & bin

I sometimes see errors with value 3, sometimes with 5 as the starting value - FWIW!

1technophile, 2018-10-08:
Thanks for pointing this, I reproduce the issue

terafin, 2018-10-08:
My pleasure! Let me know if I can help with anything!

1technophile, 2018-10-13:
@terafin @petricam, the issue has been corrected in master, if you want to test it don't hesitate

terafin, 2018-10-13:
Thank you, it works perfectly! :)  Closing!

closed 2018-10-13 by terafin