1technophile/OpenMQTTGateway #283: help wanted with time interval between ir repeat signal

savage872, 2018-10-12:
I hope someone can help with my little problem.
I'm using the openmqttgateway on a nodemcu with ir sender and receiver.
I'm also using it with homeassistant.
I use it for the tv/surround volume.
When I press volume up/down on tv remote it publishes to mqtt, home assistant then has an automation publishes another code that the nodemcu then sends to my surround sound via ir sender.
The problem that I have is when I hold down volume up or down it will send a signal every about 3 seconds and I'm trying to reduce that time somehow so I don't have to wait a while to turn volume up or down.
Any help or suggestion is appreciated!

1technophile, 2018-10-13:
OMG by default remove duplicates code to avoid pollution on the MQTT broker, you can go down the value of 3 seconds here:
https://github.com/1technophile/OpenMQTTGateway/blob/c1caf8f36e11682cadc369a5eff700488b4dc7e6/User_config.h#L122

savage872, 2018-10-13:
Thanks a lot! Is there any way of changing that value without reflashing?

1technophile, 2018-10-13:
Not at this time

closed 2018-10-13 by savage872