1technophile/OpenMQTTGateway #285: Default Sketch exceeds Memory Limits (Arduino Uno)

thehaxxa, 2018-10-18:
I have been trying to setup the Arduino Uno with Ethernet Shield for IR control, however even the base sketch exceeds the maximum memory available for the platform and is very close to exceeding the allocation for program space. 

This is with Arduino 1.8.7 running on Windows 10, using the base libraries (included with 0.8 release) with the IR Control module. Even if it could be compiled just under the threshold, stability is a concern given how little memory is available. This is the case for both 0.7 and 0.8 release. Clearly I am doing something wrong? only change made was to the User Config file to set a preferred IP address. 

`Sketch uses 29362 bytes (91%) of program storage space. Maximum is 32256 bytes.

Global variables use 2094 bytes (102%) of dynamic memory, leaving -46 bytes for local variables. Maximum is 2048 bytes.`

Compiling under 0.7 is successful but has concerningly low memory: 
`Low memory available, stability problems may occur.`

1technophile, 2018-10-18:
Strange I downloaded v0.8 and compiled for arduino UNO, here is what I get:
```
Le croquis utilise 27774 octets (86%) de l'espace de stockage de programmes. Le maximum est de 32256 octets.
Les variables globales utilisent 1697 octets (82%) de mémoire dynamique, ce qui laisse 351 octets pour les variables locales. Le maximum est de 2048 octets.
```

If you deactivate the TRACE in [user_config.h](https://github.com/1technophile/OpenMQTTGateway/blob/5d4c788e20a529589349800e285df6564752f2df/User_config.h#L133)
`#define TRACE 0  // 0= trace off 1 = trace on`

you should get:
Le croquis utilise 27240 octets (84%) de l'espace de stockage de programmes. Le maximum est de 32256 octets.
`Les variables globales utilisent 1697 octets (82%) de mémoire dynamique, ce qui laisse 351 octets pour les variables locales. Le maximum est de 2048 octets.`

If you want to go down memory occupation of global variables you have to play with:
`#define MQTT_MAX_PACKET_SIZE 256`
in [pubsubclient.h](https://github.com/1technophile/OpenMQTTGateway/blob/5d4c788e20a529589349800e285df6564752f2df/lib/pubsubclient/src/PubSubClient.h#L26)

```
Le croquis utilise 27240 octets (84%) de l'espace de stockage de programmes. Le maximum est de 32256 octets.
Les variables globales utilisent 1441 octets (70%) de mémoire dynamique, ce qui laisse 607 octets pour les variables locales. Le maximum est de 2048 octets.
```

And you should have no warning anymore

thehaxxa, 2018-10-19:
Ok, with the changes made:

'#define MQTT_MAX_PACKET_SIZE 256' & '#define TRACE 0' , I can get it to compile:

' warning: invalid conversion from 'const byte* {aka const unsigned char*}' to 'uint8_t* {aka unsigned
char*}' [-fpermissive]   Ethernet.begin(mac, ip, Dns, gateway, subnet);

Sketch uses 27128 bytes (84%) of program storage space. Maximum is 32256 bytes.
Global variables use 1709 bytes (83%) of dynamic memory, leaving 339 bytes for local variables.
Maximum is 2048 bytes.
Low memory available, stability problems may occur.

'

Still seeing the low memory warning but at least it compiles. Despite this, my MQTT output is as shown:
'
home/OpenMQTTGateway/LWT Online
home/OpenMQTTGateway/version 0.8
home/OpenMQTTGateway/LWT Offline
home/OpenMQTTGateway/LWT Online
home/OpenMQTTGateway/version 0.8
home/OpenMQTTGateway/LWT Offline
home/OpenMQTTGateway/LWT Online
home/OpenMQTTGateway/version 0.8 ...
'

Constantly switching from offline to online, is this related to low memory?

1technophile, 2018-10-19:
Yes it could, could you post your config_ir.h please ?

thehaxxa, 2018-10-19:
I haven't changed any configuration thus far, all I have done is included the default config_ir.h and simply tested with ` #define ZgatewayIR     "IR"  ` set. The same issue occurs with the RF module enabled, I am considering ordering an Arduino Mega to ensure memory issues aren't the cause.

1technophile, 2018-10-19:
yes with Mega you will be safer.

If you want to stay with Arduino Uno you can also try to go down MQTT max packet size to 128
`#define MQTT_MAX_PACKET_SIZE 128`

thehaxxa, 2018-10-20:
No luck, I did some more experimenting tonight and I definitely think I'm having memory issues: 

home/OpenMQTTGateway/LWT Offline
home/OpenMQTTGateway/LWT Online
hom/OpenMQTTGateway/version 0.8
home/OpenMQTTGateway/LWT Offline
home/OpenMQTTGateway/LWT Online
hom/OpenMQTTGateway/version 0.8
home/OpenMQTTGateway/LWT Offline

notice the home is omitted to "hom". I have purchased an Arduino Mega and new RF/IR Modules, I'll wait for their arrival and test against those, if possible keep this issue open and I'll update it when the modules arrive from Aliexpress.

thehaxxa, 2018-10-25:
I tried again using an ESP8266 (NodeMCU) today and Arduino Mega; both platforms worked as expected. I think the Arduino Uno/Nano and other ATMega328P based boards aren't optimal with OpenMQTTGateway (due to Memory Constraints). It might be a good idea to include a note in the Wiki regarding memory constraints.

1technophile, 2018-10-27:
Yes I agree, don't hesitate to add a note to the wiki

closed 2018-10-27 by thehaxxa