1technophile/OpenMQTTGateway #287: Arduino code not building

UmerFraz, 2018-10-22:
Hi. I cant get the code to complire if i follow instructions for esp32 for a btgateway plus dht and pir sensor. Get errors like "no supporting function macaddress" errors. It does compile fine if i use the latest esp32 dev env installed from arduino board manager. But then esp turns into a zombie after 6 / 8 hours and stops repoting all together. Any help/advice will be highly appreciated.

1technophile, 2018-10-23:
Please follow the issue template there is not enough information in your request to analyse the problem.

UmerFraz, 2018-11-02:
Ok sorry. BTW, a little change from the Wiki (instead of renaming the hardware package as esp32-arduino i renamed it to esp32 removing the hyphen ) sorted my issue out and i can compile the code fine now.

closed 2018-11-02 by UmerFraz