1technophile/OpenMQTTGateway #295: V0.8 Don't work with sonoff rf bridge R1 (433RFBRIDGE v2.0 mark on board)

bapt1080, 2018-11-01:
**Describe the bug**
Flash is OK
Wifi connection OK
MQTT subcribe and post OK (example : home/OpenMQTTGateway/LWT => Online)
switch is on
**Red led blink if i press a 433mhz remote but nothing on MQTT broker or serial console** (she works too)

The bridge works fine with the tasmota firmware

i try to replace tasmota to receive KAKU protocol (it's possible with RF bridge? red led don't blink with KAKU remote)

**Environment**
* OpenMQTTGateway V0.8 (flash with IDE Arduino or esptool directly with the bin file)
* RF 433 remote (works with tasmota)

1technophile, 2018-11-01:
Could you post your user_config.h without credentials ? 
Did you think on changing the rf bridge switch position after flashing ?

bapt1080, 2018-11-04:
[image]
Yes the switch is ok, changed after flashing

[User_config.txt](https://github.com/1technophile/OpenMQTTGateway/files/2546442/User_config.txt)

and the serial monitor log :

`gatewaySRFB setup done 
Serial Baud19200
MQTT_MAX_PACKET_SIZE
1024
Setup OpenMQTTGateway end
MQTT connection...
Connected to broker
Subscription OK to the subjects
Uptime (s)
120
Remaining memory
46416
RSSI
-60
SSID
WLANFB
Activated modules
SRFB`

bapt1080, 2018-11-06:
After many try still no 433 frame in MQTT. 

I can not find the code that makes the red LED blink when I press the remote control for SRFB (for RF and RF2 i find it but not for SRFB, it's "hardware" fonction? not in the ESP?)

bapt1080, 2018-11-13:
I flash the EFM8BB1 with https://github.com/Portisch/RF-Bridge-EFM8BB1

always work on tasmota and not on openMQTTgateway

1technophile, 2018-11-13:
never tested the sonoff rf bridge with this firmware do you know if it modify the way EFM8BB1 talk to the ESP?

DJBenson, 2018-11-14:
Error when compiling 0.9beta;

`Arduino: 1.8.7 (Mac OS X), Board: "Generic ESP8285 Module, 80 MHz, Flash, ck, 26 MHz, 1M (64K SPIFFS), 2, v2 Lower Memory, Disabled, None, Only Sketch, 115200"

Build options changed, rebuilding all
/Users/jonathan.thomson/Documents/GitHub/OpenMQTTGateway/ZgatewaySRFB.ino: In function 'void _rfbDecode()':
ZgatewaySRFB:130:17: error: 'repeatSRFBwMQTT' was not declared in this scope
             if (repeatSRFBwMQTT){
                 ^
exit status 1
'repeatSRFBwMQTT' was not declared in this scope

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.`

DJBenson, 2018-11-14:
I applied the fix to the Arduino project file for v0.8 and it works - not sure why 0.9beta won't compile but I'm happy with a working 0.8 build.

Thanks for the fix!

DJBenson, 2018-11-14:
Superb, thanks. And many thanks for this project!

bapt1080, 2018-11-14:
it works! thx
and with portisch on EFM8BB1 , but i cant use the advanced fonctionnality of portish with openMQTTGateway 

and KAKU is not recognized by portisch (sniffing fonction activated with tasmota don't detect KAKU remote, the led blink but nothing is send by serial between EFM8BB1 and esp :-( )

i will use rf2 with wemos, it works but my receiver is not so powerfull, i wait a new one

1technophile, 2018-11-16:
issue corrected with 0.9beta, thanks for pointing it.

I close the issue, if there is something wrong don't hesitate to reopen it

closed 2018-11-16 by 1technophile