1technophile/OpenMQTTGateway #296: Odd mi flora issue

SenexCrenshaw, 2018-11-01:
I have a mi flora (or i hope not a knock off) which connects fine to my phone and gives me data. The firmware and reported latest is 2.7.0. Not 3.1.8 which i have seen in the docs here. OpenMQTTGateway sees the device but does not send data. I get this

Get Name 
22:48:26.419 -> Flower care
Get RSSI 
22:48:26.454 -> home/OpenMQTTGateway/BTtoMQTT/C47C8D6736C5 -79
22:48:26.488 -> Get service data 
22:48:26.488 -> service_data
22:48:26.488 -> 3102980006c536678d7cc40d
22:48:26.488 -> Get service data UUID
22:48:26.488 -> 0000fe95-0000-1000-8000-00805f9b34fb
22:48:26.488 -> Processing BLE device data
BT Task running on core 0

The last mqtt message i receive is
home/OpenMQTTGateway/BTtoMQTT/C47C8D6736C5/servicedata 3102980006c536678d7cc40d

I am using OpenMQTTGateway v0.8. Any ideas?

1technophile, 2018-11-01:
Did you wait at least 15 min to see if there is no other readings with temp or moisture ?

SenexCrenshaw, 2018-11-01:
yes. This has been running for about 12 hours and i am still just getting

Get Name 
09:38:22.285 -> Flower care
09:38:22.285 -> Get RSSI 
09:38:22.285 -> home/OpenMQTTGateway/BTtoMQTT/C47C8D6736C5 -55
09:38:22.285 -> Get service data 
09:38:22.285 -> service_data
09:38:22.285 -> 3102980006c536678d7cc40d
09:38:22.285 -> Get service data UUID
09:38:22.285 -> 0000fe95-0000-1000-8000-00805f9b34fb
09:38:22.319 -> Processing BLE device data
BT Task running on core 0

home/OpenMQTTGateway/BTtoMQTT/C47C8D6736C5/name Flower care
home/OpenMQTTGateway/BTtoMQTT/C47C8D6736C5 -52
home/OpenMQTTGateway/BTtoMQTT/C47C8D6736C5/servicedata 3102980006c536678d7cc40d

I havent dug into the code but is it possible the coreTask in zgatewaybt is breaking out of the while(true) loop? It seems it displays "BT Task running on core 0" right after it tries to process the ble data.

1technophile, 2018-11-02:
what is the mi flora app you are using?

SenexCrenshaw, 2018-11-02:
It is flower care.

1technophile, 2018-11-02:
I'm using the same, and just checked again, I have 3.1.8 and the software proposes to update to 3.1.9.
What is the region you choose on the app?
I choose "Others"

SenexCrenshaw, 2018-11-02:
It was set to others. So craziest thing. I checked every day for a firmware upgrade and finally today it told me i can go to 3.1.9. I did that and i am good to go! Not sure why the app wouldnt put 3.1.8 on the device. I tried removing the device and the app multiple times before 3.1.9 came out. Thanks for you help and amazing program!

closed 2018-11-02 by SenexCrenshaw