1technophile/OpenMQTTGateway #297: TFA / Oregon Scientific Temp/Hum (or whole weather units) Sensor Support

cryptooth, 2018-11-01:
Is it possible to add/decode TFA or Oregon Scientific Tem/hum Sensor support?

1technophile, 2018-11-02:
did you try ZgatewayPilight?

cryptooth, 2018-11-02:
I just activate and upload ZgatewayPiLight. PiLight activated (in serial monitor)

When I push some RF button or open door contact, I get the some codes. Like this:

```
{"id":"B7","unit":63,"state":"off"}
qos : 0, retain : false, cmd : publish, dup : false, topic : home/OpenMQTTGateway/PilighttoMQTT/rev2_switch/B7, messageId : , length : 86, Raw payload : 1233410510034583466553444341171101051163458545144341151169711610134583411110210234125
```

But I can't get any Oregon or TFA sensor information via RF.

Any advice?

Thanky.

1technophile, 2018-11-03:
Yes could you go into pilight documentation so as to check if your sensors are compatible and see how the behave with pilight ?

cryptooth, 2018-11-16:
My TFA sensor is in pilight compatibility list. (TS34C / 30.3150) But can't get any code. Still trying...

1technophile, 2019-02-15:
I have just updated the development branch with the last version of espilight, they have added a new tfa protocol. Could you test if it works for you?

cryptooth, 2019-02-16:
I think my HM10 module corrupted. I just order new ESP32 module from Aliexpress. I will inform you after arrived.

1technophile, 2019-03-01:
I think you don't need the HM10 to do the test, just a simple RF receiver will be enough.

cryptooth, 2019-03-02:
I'm confused, sorry. I'm also dealing with xiaomi mijia BT sensors at the same time. 

I just tried again. I have not yet received any data from TFA sensors. I've updated pilight libraries on Arduino.

I'll change the RF module and try again.

1technophile, 2019-03-04:
Maybe it is not compatible.
Here is a tested [compatible weather station](https://www.banggood.com/Digoo-DG-R8S-433MHz-Wireless-Digital-Hygrometer-Thermometer-Weather-Station-Remote-Sensor-p-1139603.html?p=SZ25159101562014119R&custlinkid=135301) with ZgatewayPilight

1technophile, 2019-04-15:
I'm closing this issue, may I suggest you to add a request of integration in pilight library

closed 2019-04-15 by 1technophile