1technophile/OpenMQTTGateway #301: Support for sf-501 (flamingo) RF devices

Devqon, 2018-11-06:
**Is your feature request related to a problem? Please describe.**
This library doesn't seem to support the sf-501 switches from Flamingo.

**Describe the solution you'd like**
Support for sf-501 devices

**Describe alternatives you've considered**
There are libraries that work with the sf-501 devices, but I would like to see it integrated in this project so I can use multiple protocols.

**Additional context**
I have 3 sf-501 devices, which are controlled by a (wifi-connected) SF-501SHC smart controller. I do not like that I can only talk to the devices through the cloud, so sending the RF signals myself would be a solution.

Sources:
- SF-501: https://gathering.tweakers.net/forum/list_messages/1802639
- Arduino library: https://github.com/arjhun/arduino-sf501remote

1technophile, 2018-11-10:
Did you try ZgatewayPilight from the master branch?
This gateawy is supporting more devices compared to ZgatewayRF.
Another possibility is to try a new version of ZgatewayRF from @animavitis that you can find here:
https://github.com/animavitis/OpenMQTTGateway/tree/rcswitch-update

Devqon, 2018-11-16:
I haven't got my ethershield yet so I cannot test yet, but it seems that the Flamingo switches use the NewKaku protocol, which is also supported by this library. Therefore my guess is that the Flamingo switches will work :)

1technophile, 2019-06-21:
I'm closing this issue, if you need other info don't hesitate to reply

closed 2019-06-21 by 1technophile