1technophile/OpenMQTTGateway #302: Fefature: add write characteristic to BLE

animavitis, 2018-11-08:
**Is your feature request related to a problem? Please describe.**
I want to be able to control BLE thermostat from MQTT

**Describe the solution you'd like**
Send from HA MQTT message:
`{ 'action': 'writeCharacteristic', 'uuid': '47e9ee2b-47e9-11e4-8939-164230d1df67', 'value': [ 128,  40, 128, 128, 128, 128, 128 ] }`

in result OMG send write chatacteristic to uuid and send confirmation message

**Describe alternatives you've considered**
https://github.com/seidler2547/ble-mqtt-bridge this is ready and working solution, but it need to be run from my HA server and not all thermostats are in BLE range

fightforlife, 2020-11-09:
Did anything happen here in the meantime?

1technophile, 2020-11-09:
> Did anything happen here in the meantime?

We are now able to connect to BLE devices ([LYWSD03MMC](https://github.com/1technophile/OpenMQTTGateway/blob/a184355f9bb03bb59fd67197f4ddfba184d552b0/main/ZgatewayBT.ino#L539)) so we are closer to the implementation of this kind of functionality, but nobody did it yet.

DigiH, 2022-02-06:
@animavitis @fightforlife Isn't what you were looking for what was introduced in 0.9.7 and extended with the **immediate** parameter in 0.9.9?

https://docs.openmqttgateway.com/use/ble.html#read-write-ble-characteristics-over-mqtt-esp32-only

1technophile, 2023-06-12:
Done, feel free to reopen if any questions

closed 2023-06-12 by 1technophile