1technophile/OpenMQTTGateway #306: General Question and Issue: AM312 possible replacement for HC-SR501?

djraidar, 2018-11-10:
Hi!
I was wondering if I can use an AM312 instead of the HC-SR501 for motion detection.

And, the issue:
I am using the ESP32 with a DHT22 and a Motion Sensor. In the serial monitor, it keeps looping 'creating hcsr501 buffer' hundreds of times, barring any info from BT or DHT info, any insight why? Could it be me not using 5V power and trying to operate this from the 3.3v pin of the ESP32?

1technophile, 2018-11-10:
Hi,

maybe try to put further the motion sensor, they can causes perturbation to the chip.

thewillwilson, 2019-03-18:
I've also got this issue.

chesterbr, 2019-08-13:
Interestingly, I just tried the HC-SR501 and got the same output:

```
23:13:16.201 -> Creating HCSR501 buffer
23:13:16.201 -> Pub json into:
23:13:16.201 -> home/OpenMQTTGateway/HCSR501toMQTT
23:13:16.237 -> {"hcsr501":"true"}
23:13:16.237 -> Creating HCSR501 buffer
23:13:16.237 -> Creating HCSR501 buffer
23:13:16.237 -> Creating HCSR501 buffer
23:13:16.237 -> Creating HCSR501 buffer
23:13:16.237 -> Creating HCSR501 buffer
23:13:16.237 -> Creating HCSR501 buffer
23:13:16.237 -> Creating HCSR501 buffer
23:13:16.237 -> Creating HCSR501 buffer
23:13:16.237 -> Creating HCSR501 buffer
23:13:16.237 -> Creating HCSR501 buffer
23:13:16.237 -> Creating HCSR501 buffer
...
```

- Even if I unplug the sensor, I get the loop (about 462 messages per second, in an ESP8266).
- The message on the top of the log was indeed published to the gateway, which works otherwise in the other functions (RF and IR <-> MQTT, in my case). 

I have uncommented the `#define ZsensorHCSR501` as described in https://github.com/1technophile/OpenMQTTGateway/wiki/User-guide-PIR (the `#include` doesn't need to be uncommented, because it is [gated](https://github.com/1technophile/OpenMQTTGateway/blob/ca7d5d4195488552025087960709a4e8443d78b0/main/main.ino#L72-L74) by the define, I'll submit a PR for the docs as I tinker a bit more with this.

chesterbr, 2019-08-13:
Oh, looking again at [the code](https://github.com/1technophile/OpenMQTTGateway/blob/development/main/config_HCSR501.h#L38-L39), I see that I should have plugged it to D5 (for an ESP8266), not a D7 (seems like another patch to the docs is upcoming).

But most interesting is that it *works*: I get a `home/OpenMQTTGateway/HCSR501toMQTT {"hcsr501":"true"}` when there is movement, and `false` after a few seconds leaving the sensor alone. So the only bug here is the spamming of "Creating HCSR501 buffer" messages, let me look at that...

closed 2019-08-13 by 1technophile

Kinetik, 2019-11-15:
Put the HCSR501 on 5v (and for data D5 pin) and the DHT sensor on 3v3 (for data D2 pin). That should do it. Cheers

Keima645, 2021-02-01:
HCSSR501 is much more susceptible to Wifi Noises than AM312. You might want to consider that as a better alternative

PR #442 (2019-08-13): Avoid spam of buffer message in hcsr501
Hello, this is my first code PR on this (awesome) project!

It (hopefully) fixes #306 - in which some users noticed that the HCSR501 module prints the `Creating HCSR501 buffer` message hundreds of times per second (despite working, as I wrote in that issue's comments).

### The changes

It does so by removing the line that prints the message. It also adds a message for when the movement starts (mostly to match when the movement stops, can remove that or both if those aren't necessary).

### Testing

My logs on an ESP8266, before the fix:

```
...
23:13:3 [... cut]