1technophile/OpenMQTTGateway #307: Cannot compile OpenMQTTGateway 0.8

arnold-maderthaner, 2018-11-10:
I tried to follow the wiki entry but I got stuck with the following:

Invalid library found in /Users/arnoldmaderthaner/Documents/Sketchbooks/hardware/espressif/arduino-esp32/libraries/AzureIoT: /Users/arnoldmaderthaner/Documents/Sketchbooks/hardware/espressif/arduino-esp32/libraries/AzureIoT

OpenMQTTGateway:60:26: error: PubSubClient.h: No such file or directory
compilation terminated.
exit status 1
PubSubClient.h: No such file or directory

how can I fix that ?

Thanks !

1technophile, 2018-11-10:
Hello,

You should have missing libraries, did you copied:
OpenMQTTGateway/lib to "sketchbook folder"/libraries
?

arnie580, 2018-11-13:
Hi, I am having a similar issue, I think to be honest I've just put the files in the wrong place but can't work out the right place!

1technophile, 2018-11-13:
yes

1technophile, 2018-11-16:
I close this issue, if you canno't succeed in your compilation don't hesitate to go to the [forum](https://community.openmqttgateway.com)

closed 2018-11-16 by 1technophile

keteflips, 2019-01-02:
Just copy PubSubClient.h from:

.../libraries/pubsubclient/src//home/keteflips/sketchbook/libraries/pubsubclient/src/PubSubClient.h 

to

.../libraries/pubsubclient/src//home/keteflips/sketchbook/libraries/pubsubclient/PubSubClient.h