1technophile/OpenMQTTGateway #308: Change default will_Message & Gateway_AnnouncementMsg to lowercase

animavitis, 2018-11-11:
Before submitting a problem please check the troubleshooting section
https://github.com/1technophile/OpenMQTTGateway/wiki/Troubleshooting

**Describe the bug**
Default availability_payload & payload_not_available in Home Assistant is "online" & "offline" (lowercase). Now i have to add to all senors/binary_sensors/switches/lights 2 additional config lines for each. ie:
`  - platform: mqtt
    state_topic: "home/salon/433toMQTT"
    availability_topic: "home/salon/LWT"
    name: "sypialnia - okno"
    unique_id: 'syp_okno'
    device_class: opening
    value_template: '{{ value_json.value }}'
    payload_on: "3886858"
    payload_off: "3886862"
    availability_payload: "Online"
    payload_not_available: "Offline"
    optimistic: false
    retain: true`
After change it will be:
`  - platform: mqtt
    state_topic: "home/salon/433toMQTT"
    availability_topic: "home/salon/LWT"
    name: "sypialnia - okno"
    unique_id: 'syp_okno'
    device_class: opening
    value_template: '{{ value_json.value }}'
    payload_on: "3886858"
    payload_off: "3886862"
    optimistic: false
    retain: true`

**Environment (please complete the following information):**
* OpenMQTTGateway 0.9beta

* You should not have a compilation error if you use the versions of the libraries linked into the libraries folder, this badges show you the state of the compilation 
[image]
* If you are not sure this is a bug or an enhancement post your question to the forum below
[image]

1technophile, 2018-11-11:
Sorry but I don't see exactly what you are expecting, are you talking about having automatic discovery into HASS?

animavitis, 2018-11-13:
yes, i want to have hass autodiscovery, but for this i need configuration json to be as short as possible. 
After simply changes: "Online" to "online" & "offline" to "offline"is possible to remove properties from each sensor configuration: availability_payload: "Online" payload_not_available: "Offline".

1technophile, 2018-11-16:
Hello, as it is done in your PR I think we can close this issue

closed 2018-11-16 by 1technophile