1technophile/OpenMQTTGateway #318: tests v0.9beta

1technophile, 2018-11-25:
Non regression testing of v0.9beta:

Here is below the functions to test
https://docs.google.com/spreadsheets/d/1_5fQjAixzRtepkykmL-3uN3G5bLfQ0zMajM9OBZ1bx0/edit?pli=1#gid=1617051124

Take the code from the master branch

1technophile, 2018-11-27:
thanks, missing macro added

kegra, 2018-11-30:
hi , i used ZgatewayBT and ZmqttDiscovery , trying to compile give me error
ZmqttDiscovery:42: error: 'BME' was not declared in this scope
                   BME, "tempc", getUniqueId("bme", "tempc"),
                   ^
exit status 1
'BME' was not declared in this scope

after commenting //#define discBme280 in ZmqttDiscovery i got it compiled correctly.

Now on the operation side , i had already used v08 and was getting quite nice updates from my Miflora  inside home assistant but in V9 i never get all the updates from all 4 sensors (tem ,moi, lux, fer), i have tried both json and simple topic per sensor . I let it run for few hours but i always was getting maximum two of them and never the other two. I was getting session data but don't know if this  is usable and how. i was monitoring all subjects /home/# so this has nothing to do with my configuration inside home assistant.

1technophile, 2018-11-30:
Thanks for the feedback, as mqtt discovery is only implemented for BME at this time I should add some macros to avoid the error you have seen.

Regarding BLE did you changed something in the [config_BT.h](https://github.com/1technophile/OpenMQTTGateway/blob/master/config_BT.h) between both versions?

kegra, 2018-11-30:
i have exactly the same parameters inside the  config_BT.h nothing changed i even tried different scan intervals 10, 20 , 30 seconds i have both Miflora and the BLE temp and humidity sensor, the funny thing is in v8 i don't get updates from the temp and humidity sensor but i get them perfect in the V9 but miflora does not work in v9 at least for me. Have you tried both as the same time is there any strange conflict ? (maybe stupid of me to say this ) but have seen crazy stuff happening all those years.

1technophile, 2018-11-30:
Yes I tried mi flora and mi jia at the same time but from what you are saying I need to retry ;-)

kegra, 2018-11-30:
do you have any recommendations i can try ? maybe now i am thinking the BLE gateway (xiaomi camera acting as ble gateway) causes a trouble ? (if i am hijaking with irrelevant stuff i can continue to another thread or to the openmqttgateway community page if you like)

1technophile, 2018-11-30:
could you just let ON your mi floras and mi jias?

kegra, 2018-11-30:
i will leave this today all night to check again. thank you for your support and for this great project !

kegra, 2018-11-30:
quick update, everything seems to work normal now, after tweaking a bit , but most important i think the BLE gateway (xiaomi cam)  was causing the trouble , will test further after turning on the BLE Gateway again. thank you.

kegra, 2018-12-04:
V9 seems solid for

Mi jia temperature, battery and humidity sensor integration  OK
Xiaomi flora (flower care) integration with HM10 OK
add MQTT publishing of service data from BLE devices OK
Web portal for wifi and mqtt configuration (ESP8266) OK
Don't go to AP mode if wifi is lost OK
send BLE beacons mac adress and rssi to MQTT OK

jeanniquini, 2018-12-06:
@1technophile about issue [#241](https://github.com/1technophile/OpenMQTTGateway/issues/241)
I'm using RF + BLE + DHT modules with nodemcu. And randomly openmqttgateway restarts. I am using to track a mi band 3 but as I have a google home mini, it also appears in the scans. With RF + DHT, I did not have this problem.

jeanniquini, 2018-12-06:
If comment #define subjectHomePresence Base_Topic "home_presence/" Gateway_Name
I've got:
'subjectHomePresence' was not declared in this scope

kegra, 2018-12-06:
Today i am trying to use the home presence publishing , i get JSON formatted data under the topic
home/home_presence/OpenMQTTGateway/
{
  "id": "",
  "rssi": -65,
  "distance": 1.987855
}

the issue is i don't get anything under the id so i do not see how can i identify different BLE devices from here ?

jeanniquini, 2018-12-07:
@kegra I am using rssi posted on specific mac. e.g.: home / OpenMQTTGateway / BTtoMQTT / 'MAC ADRESS'

kegra, 2018-12-07:
@jeanniquini i can do this also using the topic as you posted using /distance instead of rssi but the thing is that the id should be included in the json for home presence. i was using exactly what you propose in the V.0.8 but in V.0.9 distance and home presence is much better for this purpose.

jeanniquini, 2018-12-08:
@kegra are you using RF + BLE together? OMG is randomly crashing here.

kegra, 2018-12-08:
@jeanniquini unfortunately i don't , i only use BLE.

jeanniquini, 2018-12-08:
@1technophile 
All right. I have the newer firmware version. Just trying to figure out what's causing the OMG crash... I turned off the BLE and everything is normal. I want to disable the service data to see if the problem is there.

1technophile, 2018-12-08:
You are having a watchdog wdt reset ?

jeanniquini, 2018-12-08:
The RF module stops working after some time and the DHT delays the sending of the data a few times.

jeanniquini, 2018-12-08:
Okay! I'll test for a few hours and come back here.

Dilbert66, 2018-12-09:
I had to comment 'include "wifi.h" in user_config.h in order to compile when using esp8266 2.4.2 board config.  Kept giving me an error about a missing write_p function.

1technophile, 2018-12-09:
@dilbert66 you should have a wrong library version on your environment. Indeed the build is working well for all boards here : 
https://travis-ci.org/1technophile/OpenMQTTGateway

Dilbert66, 2018-12-09:
Yes you are correct. I see my issue. Thank you.

jeanniquini, 2018-12-10:
@1technophile 
Keep restarting. I did several tests (firmware < v601, > v601, with and without JSON, different reading times). And I realized that when the service data is too large, the message is split into two, doubling the time set in the settings. Would be possible for an option with firmware > v601 but discarding the service data? 

`home/OpenMQTTGateway/BTtoMQTT/CDEDE7B9F4D2
{"rssi":-46,"servicedata":"00924ae67998ee5a6bbb2287e312e759c702cdede7b9f4d2"}
4f4b2b444953413a95a5b62dea6403df0a09ffe0001640ca7ec6480d0a
4f4b2b444953413a95a5b62dea6403df1c03039ffe17169ffe00000000000000000000000000000000000000000d0a
Creating BLE buffer
`

45s
[image]

jeanniquini, 2018-12-11:
Confirmed. It's my Mi Band service data ... I've left it out of range and the OMG has been 13 hours without crash so far.

1technophile, 2018-12-12:
Thanks for the info, I will add an option to don't publish service data

jeanniquini, 2018-12-19:
And crashing.

1technophile, 2018-12-20:
Hello,

So as to have a view of the remaining test for v0.9beta you can lean on this file:
https://docs.google.com/spreadsheets/d/1_5fQjAixzRtepkykmL-3uN3G5bLfQ0zMajM9OBZ1bx0/edit?pli=1#gid=0
Mainly we are missing:
- IR gateway tests
- Tests of OMG with arduino mega 
- and some other functions

If you can help for these tests don't hesitate to go and give your feedback

1technophile, 2018-12-20:
@jeanniquini do you see always the same string size when it crashs?

jeanniquini, 2018-12-20:
@1technophile 
I did not check the logs until it crashed. I only noticed that after a few minutes after publishing these duplicate and similar messages it crashes. But I can do that.
Did you see that the service data is not published but it appears in the serial log?

1technophile, 2018-12-20:
Yes i deactivated the publishing only. If you think this is due to traces, you can deactivate traces by commenting the corresponding macro.
https://github.com/1technophile/OpenMQTTGateway/blob/3d459e058277a88582f07148607c5f855041a268/User_config.h#L172

jeanniquini, 2018-12-20:
I tested it now. Trace is not the problem.

1technophile, 2018-12-20:
Ok thanks, could you say me in this case if there is a constant limited number of characters ?

jeanniquini, 2018-12-20:
Always 252 characters.

1technophile, 2018-12-22:
@jeanniquini I ve added a limit to the servive data string size, could you test it to see if it still crash.
https://github.com/1technophile/OpenMQTTGateway/blob/Servicedatalimit/ZgatewayBT.ino

jeanniquini, 2018-12-23:
RF stopped working after some time. Now mqtt messages are not duplicate but take double or triple the time set. Few times at the correct time. I set 60s to test. If I reset the nodemcu, RF returns to work correctly.

jeanniquini, 2018-12-24:
It was not crashing, only the rf stopped working. After resolding the module, the rf is ok. But the BLE does not follow the time set between the readings.

keschner, 2019-01-08:
@1technophile I switched from 0.8.1 to 0.9beta because the BT data for Mi flora were not read correctly in 0.8. In 0.9 this works perfectly with my HM-11, but RF stops working. The RF command is published, but nothing is send. I've checked the configured RF emitter pin because it's changed in dev-branch to 4, but in master it's still 3, so my hardware setup should work without changes (I've also tried with pin 4 :wink:). All lib's are updated and I didn't change anything in the RF config. If I switch back to 0.8.1 RF is working.

Do you have a hint what could cause this issue?

1technophile, 2019-01-08:
@keschner thanks for your feedback ! 
Could you point me what are you getting in the serial monitor when you are testing RF with 0.9beta ? 
Would it be possible to test the v0.9beta with the v0.8 rcswitch library included ? 
The library changed between both versions

keschner, 2019-01-08:
@1technophile thanks for your fast response. To check the serial output I've changed the emitter pin to 4, so I can use the RX pin for the serial monitor.

With this changed setup I've checked the serial output and get the following error (74880 Baud not the configured 115200 :confused:):
`Fatal exception (28): 
epc1=0x40001800, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00007ff0, depc=0x00000000`

With the unchanged rcswitch lib I don't get this error. But it doesn't work either. I only get 
`Hey I got a callback` in the serial monitor.

1technophile, 2019-01-08:
Hmm, I see, please go back to the v0.9beta full setup (with the v0.9beta rcswitch library)

Your issue should be that simplePublishing is deactivated by default with v0.9beta. And you should publish your commands with simplePublishing.
Try to uncomment it in [user_config.h](https://github.com/1technophile/OpenMQTTGateway/blob/c7fe43eb97b5de850acf312fa6bf31a54b05cddc/User_config.h#L126) it should help you

keschner, 2019-01-08:
:man_facepalming: ... haven't seen that. Thanks for your help. Now everything works as expected.

1technophile, 2019-01-17:
Hello, could you indicates what do you want to achieve?

jeanniquini, 2019-01-21:
@1technophile Any progress in combining RF + BLE?  The limit to the service data string size prevent the crashes but the RF stops working after a while.

1technophile, 2019-01-31:
@jeanniquini not yet, I will open a separate issue

closed 2019-01-31 by 1technophile