1technophile/OpenMQTTGateway #321: 'BME' was not declared in this scope

bereska, 2018-12-02:
trying to flash OpenMQTTGateway.ino to my ESP8266 I get the following error:

/Users/a0000/Downloads/OpenMQTTGateway/ZmqttDiscovery.ino: In function 'void pubMqttDiscovery()':
ZmqttDiscovery:42:19: error: 'BME' was not declared in this scope
                   BME, "tempc", getUniqueId("bme", "tempc"),
                   ^
exit status 1
'BME' was not declared in this scope

any help is appreciated
thanks

1technophile, 2018-12-02:
If it is not for testing you should take the v0.8 version and not the v0.9beta

bereska, 2018-12-02:
thanks for your prompt reply
stupid me, i should have known what I was downloading
0.8 works
1543782382: New client connected from 192.168.2.58 as OpenMQTTGateway (c1, k15, u'openmqtt').

quick question: I saw other people asking about Oregon and Ea2 temperature sensors support, any progress on that?
thank you

bereska, 2018-12-02:
this is what I'm getting on serial monitor:

Uptime (s)
1440
Remaining memory
43608
RSSI
-79
SSID
bereska24
Activated modules
RFIRBT

does it mean I have no supported 433Mhz devices around?
thanks

1technophile, 2018-12-03:
Rf 433mhz temperature sensors are not supported by v0.8

With v0.9 pilight library has been introduced, with this library some temperature sensors should be supported. You can try to redownload the master, i have corrected the BME compilation issue. Uncomment all the gateways except ZgatewayPilight and try to see if you get signals from your weather station.

bereska, 2018-12-03:
great!
unfortunately my sensors are not on pilight list of supported devices
let me try that anyways and report back
thank you

1technophile, 2018-12-16:
I close the issue, don't hesitate to give feedback on the community forum

closed 2018-12-16 by 1technophile