1technophile/OpenMQTTGateway #322: Question about RFM69 usage for weather station using FSK ?

mr-sneezy, 2018-12-06:
I have an RFM69 dev board and some ESP8266 modules (NodeMCU) and I'd like to try write some code to using it with OpenMQTTGateway to  receive weather station packets and send onward to HomeAssistant (so RX only). I will have to decode the packets using my own Arduino coding, and feed those values for the MQTT routines...
I should point out that this style of thing had been done many times for older weather stations sending the OOK siganlling at 433/915Mhz, but to my knowledge has never been done for the newer gen FSK transmission weather stations (li.e. the WH2900).

My question, is the RFM69 support available in OpenMQTTGateway limited to only using node/clients on a preset fixed modulation and data rate, or is there a config file that allows me to use any modulation and bandwidth etc available in the RFM69, to match what my weather station is sending and obtain the hex packets  ?

Hope the question makes sense...

1technophile, 2019-05-04:
> My question, is the RFM69 support available in OpenMQTTGateway limited to only using node/clients on a preset fixed modulation and data rate, or is there a config file that allows me to use any [...]

Hello,
Maybe you could investigate on RFM69 library repo here is the link:
https://github.com/LowPowerLab/RFM69/tree/master/Examples

1technophile, 2019-06-21:
I'm closing this issue, if you need other info don't hesitate to reply

1technophile, 2019-06-21:
May I add that with the advised receiver and emitter we can now use pilight, pilight support weather sensors like digoo dg-r8s

closed 2019-06-21 by 1technophile