1technophile/OpenMQTTGateway #325: unstable 433 rf stops sending

wolass, 2018-12-08:
**Describe the bug**
After using the OMG for a while (1 month) I observe that sending the MQTT data to the gateway stops working and it is not sending it through the RF433 Mhz Radio.

In the beginning, everything works fine but after some time the RF433 communication stops sending RF signals. After this occurs, a restart through pushing the reset button on the WEMOS D1 mini board RF433 signals (that were previously sent to the OMG) are executed. 

**To Reproduce**
Steps to reproduce the behavior:
1. Burn OMG to a WEMOS D1 mini with an RF433 radio (superheterodyne)
2. Wait for some time with normal usage: sending/ rec about 100 signals a day.
3. Observe that OMG is no longer sending the RF433 commands. (does it hang??) I do not have a serial monitor hooked up to it for such a long time. 

**Expected behavior**
I would expect it to work flawlessly. - Another question here - would it be possible to implement some kind of a watchdog that would restart the chip in case of such behavior? 

**Environment (please complete the following information):**
* OpenMQTTGateway version used (V0.8, master)
* Libraries according to cloning with submodules

1technophile, 2018-12-08:
Could you gave us more information please:

- Does the gateway disconnect from your broker (offline status on the topic "home/OpenMQTTGateway/LWT")  or even the wifi, if you try to ping it does it respond?

- Did you tested it on another board?

- Ultimately could you connect it to a serial monitor and gave us the traces before the hang?

wolass, 2018-12-09:
I will perform these three tests tommorow and report back. The problem is
that these "hangs" are not that frequent but yesterday it happened again.
After pushing the reset button everything jumps back to normal.

1technophile, 2019-05-04:
Hello,

I close the issue, if you have feedback to add don't hesitate to put a new comment

closed 2019-05-04 by 1technophile