1technophile/OpenMQTTGateway #329: Add CC1101 gatewat (RF gateway with merged rcswitch and newremoteswitch)

1technophile, 2018-12-15:
**Is your feature request related to a problem? Please describe.**
Enable to use rcswitch and newremoteswitch in the same time for sending and receiving

**Describe the solution you'd like**
Add a new gateway compatible with CC1101

https://github.com/LSatan/RCSwitch-CC1101-Driver-Lib

JensVanhooydonck, 2019-05-10:
This would also support the 868MHz devices?

1technophile, 2019-05-12:
At least it supports 868mhz, regarding which device it supports it will depend on the protocols used by these devices.

orrpan, 2019-06-18:
If you have two cc1101, wouldn't it be possible (when implemented in OMG) to use one for 433mhz and one for 868mhz? Tested a little with one as a culfw (nano), works like a charm on both frequencies.

1technophile, 2019-06-30:
Seeing the number of pins required by cc1101 I'm not sure we can have 2 working with one ESP board.
The difficulty could be to find enough operational/suitable pins

JLFN, 2019-07-25:
I found this complete prove of concept.
http://www.marco-maniscalco.de/?p=2217

1technophile, 2019-07-25:
Thanks for the info.

I'm asking myself if investing time in cc1101 is a good thing, indeed as now we have Pilight integrated and with it a wide range of RF device supported maybe investing time in pilight will have better returns.

fgather, 2019-08-15:
Interesting, look at what I did some years ago:
https://github.com/fgather/OpenMQTTGateway 
;)

orrpan, 2019-08-26:
https://github.com/orrpan/OpenMQTTGateway/tree/cc1101_rf2 
Works well for RF2, using RCSwitch-CC1101-Driver-Lib – by LSatan.

JLFN, 2019-08-26:
@orrpan Can you make it for 868mhz also?

orrpan, 2019-08-26:
> @orrpan Can you make it for 868mhz also? Are you using the same pinout as LSatan ?

I can't test 868mhz, no devices.  I've added config into config_RF.h

Yes same pinout:
* Lsatan: https://github.com/LSatan/RCSwitch-CC1101-Driver-Lib/blob/master/WIRING%20ESP8266.txt
* config_RF.h for openmqttgateway
RF_RECEIVER_PIN 4
RF_EMITTER_PIN 5

(You also need to clone/install his repository into your /..../Arduino/libraries folder)

orrpan, 2019-11-28:
@JLFN did it work with 868mhz?

AlmightyFrog, 2019-11-28:
In theory I could test 868 as I still have an (unused) diy CUL-Stick with 868 module connected.
My interest would only be in read out some w-mbus messages, but as far as I got https://github.com/puuu/ESPiLight there only is the 433 devices supported.

Am I wrong here, where is supported devices list?

1technophile, 2019-11-29:
For pilight : 
https://manual.pilight.org/protocols/433.92/index.html

For rcswitch : 
https://github.com/sui77/rc-switch/wiki

AlmightyFrog, 2019-11-29:
@1technophile please forgive me if I am wrong, but at piilight it says there "433.92Mhz" as headline so I would assume it only supports that. Not sure where the 868 are there, but anyhow on ESPiLight it says `This Arduino library is a port of the pilight 433.92MHz protocols to the Arduino platform.`

At rc-switch wiki it says `You need is an Arduino, ESP8266, etc, a 315/433MHz transmitter` which also means no 868 support.

orrpan, 2019-11-29:
Okay, pilight will not work. But I was meaning using rf or rf2 and my branch with a cc1101. I use it for 433mhz and it works flawless.

1technophile, 2019-11-29:
> @1technophile please forgive me if I am wrong, but at piilight it says there "433.92Mhz" as headline so I would assume it only supports that. Not sure where the 868 are there, but anyhow on [...]

Indeed 868mhz is not supported by RF nor Pilight.

AlmightyFrog, 2020-04-24:
I just saw PR #585 and put it on a nodemcu-32s board to test.
Have it only running since one hour so not sure how stable it is, but so far everything works fine.
Thanks for the link to https://github.com/LSatan/SmartRC-CC1101-Driver-Lib#wiring.
I already had that wired up so it was fine. But maybe in documentation it should be clearly pointed out, that default pins for GD00 and GD02 are as per config_RF.h D12 and D27.

orrpan, 2020-04-24:
The docs are updated in dev branch, not sure when it will be updated

1technophile, 2020-04-24:
https://docs.openmqttgateway.com is updated at each release for info.
I will publish a beta release in the following week.

1technophile, 2020-04-25:
I'm closing this issue as a PR #585 was done regarding this subject

closed 2020-04-25 by 1technophile