1technophile/OpenMQTTGateway #331: BLE Whitelist.

maxi1134, 2018-12-19:
**Is your feature request related to a problem? Please describe.**
A Whitelist for the Bluetooth Scan for MQTT presence location.

**Describe the solution you'd like**
A whitelist, this would allow to reduce irrelevant data on the MQTT server, Living in an appartment block, I usually get 20 to 30 devices detected per scan. 

**Describe alternatives you've considered**
A black list could also be usefull.

**Additional context**
Pretty straightforward.

1technophile, 2018-12-20:
Do you think an mqtt json payload with the list of whitelist & blacklist devices answer the need?

rickitaly, 2019-01-20:
Agree, would be great to scan and get mqtt msg only for relevant BLE devices!

tunip, 2019-01-20:
Thats the way ESPHome works. Define MAC for relevant devices. Would be great OMG has this feature too.

NogNog87, 2019-02-19:
It would be great to get a scan of one or more g-tags (beacons) that you define yourself (whitelist). Perfect would be information that the beacon is in range or not. If it not in range you get the payload "off" or similar

1technophile, 2019-02-19:
I'm planning to add this feature to the next release. Nevertheless I'm still thinking about how to set this whitelist. @maxi1134 the drawback with hardcoded list is that users that take directly the binary file will not be able to use this feature. Why would you not be interested in passing the list by mqtt?

maxi1134, 2019-02-19:
@1technophile I'm not against the idea. MQTT Would be a more elegant solution now that I think about it.
Easier to append more devices without reflashing everything.

1technophile, 2019-02-19:
ok thanks for the feedback I will design it like that

1technophile, 2019-03-18:
60f68d00f4dbf0acb629e11c5f42485228084e9a

closed 2019-03-18 by 1technophile

NogNog87, 2019-03-19:
Hello,

where i can set the Whitelist?

Greetz
Daniel

1technophile, 2019-03-19:
Hello, the infos were on the commit msg:
set black list
`mosquitto_pub -t home/OpenMQTTGateway/commands/MQTTtoBT/set -m '{"black-list":["012314551615","4C6577889C79","edfrgtyjulik"]}'`
set white list
`mosquitto_pub -t home/OpenMQTTGateway/commands/MQTTtoBT/set -m '{"white-list":["012314551615","4C65A5553C79","edfrgtyjulik"]}'`

maxi1134, 2019-05-07:
Question, would it be possible to have a separate topic for the Whitelist? This way you can update whitelist for multiple gateways at once. Myself I use 8 to do per room location.

1technophile, 2019-05-09:
Did you try to publish to all the gateways like that:
mosquitto_pub -t home/+/commands/MQTTtoBT/set -m '{"white-list":["012314551615","4C65A5553C79","edfrgtyjulik"]}'

jamesyym, 2019-06-18:
> Hello, the infos were on the commit msg:
> set black list
> `mosquitto_pub -t home/OpenMQTTGateway/commands/MQTTtoBT/set -m '{"black-list":["012314551615","4C6577889C79","edfrgtyjulik"]}'`
> set white [...]

Hi,
Could you please explain more about this, where and how to set?
Thanks a lot.

1technophile, 2019-06-18:
Hi, since this I have updated the wiki : 
https://github.com/1technophile/OpenMQTTGateway/wiki/User-guide-bluetooth-BLE-09#setting-a-white-or-black-list

maxi1134, 2019-06-20:
Hey 1Technophile.

Just updated my tester to the latest version of OMG.
Then I published 
`{"white-list":["e01126898417"]}` to  `home/Bedroom2/commands/MQTTtoBT/set`

This is the result on the serial monitor.
```
12:15:50.935 -> Hey I got a callback 
12:15:50.935 -> MQTTtoBT json set
12:15:50.935 -> WL set
12:15:50.935 -> E01126898417
12:15:50.935 -> 0
12:15:50.935 -> 1
12:15:50.935 -> 0
12:15:50.935 -> e01126898417
12:15:50.935 -> 0
12:15:50.935 -> 1
12:15:50.935 -> 0
``` 

The problem is that it keeps detecting other BL devices and publishing them.

1technophile, 2019-06-20:
Hey,

Could you post an extract of the serial monitor when publishing non expectedBLE devices?

maxi1134, 2019-06-20:
Of course,

```
15:46:52.409 -> Hey I got a callback 
15:46:52.409 -> MQTTtoBT json set
15:46:52.409 -> WL set
15:46:52.409 -> E01126898417
15:46:52.409 -> 0
15:46:52.409 -> 1
15:46:52.409 -> 0
15:46:52.479 -> Creating BLE buffer
15:46:52.479 -> BLErssi
15:46:52.479 -> -90
15:46:52.479 -> txPower
15:46:52.479 -> -59
15:46:52.479 -> BLE DISTANCE :
15:46:52.479 -> 23.44
15:46:52.479 -> Pub json into:
15:46:52.479 -> home/home_presence/Bedroom
15:46:52.479 -> {"id":"68:2e:ed:07:11:05","manufacturerdata":"⸮","rssi":-90,"distance":23.44428}
15:46:52.479 -> Get services data :
15:46:52.479 -> 1
15:47:00.574 -> BT Task running on core 0
15:47:05.812 -> Creating BLE buffer
15:47:05.812 -> BLErssi
15:47:05.812 -> -89
15:47:05.812 -> txPower
15:47:05.812 -> -59
15:47:05.812 -> BLE DISTANCE :
15:47:05.812 -> 21.52
15:47:05.812 -> Pub json into:
15:47:05.812 -> home/home_presence/Bedroom
15:47:05.812 -> {"id":"79:f6:51:f2:ee:95","manufacturerdata":"⸮","rssi":-89,"distance":21.51847}
15:47:05.812 -> Get services data :
15:47:05.812 -> 1
15:47:06.021 -> Creating BLE buffer
15:47:06.021 -> BLErssi
15:47:06.021 -> -45
15:47:06.021 -> txPower
15:47:06.021 -> -59
15:47:06.021 -> BLE DISTANCE :
15:47:06.021 -> 0.07
15:47:06.021 -> Pub json into:
15:47:06.056 -> home/home_presence/Bedroom
15:47:06.056 -> {"id":"39:32:e4:f8:a1:8d","manufacturerdata":"","rssi":-45,"distance":0.06662}
15:47:06.091 -> Creating BLE buffer
15:47:06.091 -> BLErssi
15:47:06.091 -> -65
15:47:06.091 -> txPower
15:47:06.091 -> -59
15:47:06.091 -> BLE DISTANCE :
15:47:06.091 -> 2.01
15:47:06.091 -> Pub json into:
15:47:06.091 -> home/home_presence/Bedroom
15:47:06.091 -> {"id":"45:06:27:58:2b:fd","manufacturerdata":"⸮","rssi":-65,"distance":2.009448}
15:47:06.091 -> Get services data :
15:47:06.091 -> 1
15:47:06.161 -> Creating BLE buffer
15:47:06.196 -> BLErssi
15:47:06.196 -> -81
15:47:06.196 -> txPower
15:47:06.196 -> 4
15:47:06.196 -> BLE DISTANCE :
15:47:06.196 -> ovf
15:47:06.196 -> Pub json into:
15:47:06.196 -> home/home_presence/Bedroom
15:47:06.196 -> {"id":"74:f6:1c:03:e5:eb","name":"HTC BS EBF799","rssi":-81,"txpower":4,"distance":1.159445e13}
15:47:06.300 -> Creating BLE buffer
15:47:06.300 -> BLErssi
15:47:06.300 -> -77
15:47:06.300 -> txPower
15:47:06.300 -> -59
15:47:06.300 -> BLE DISTANCE :
15:47:06.300 -> 7.12
15:47:06.300 -> Pub json into:
15:47:06.300 -> home/home_presence/Bedroom
15:47:06.300 -> {"id":"76:f1:92:c6:ed:ce","manufacturerdata":"⸮","rssi":-77,"distance":7.119795}
15:47:06.300 -> Get services data :
15:47:06.300 -> 1
15:47:06.923 -> Creating BLE buffer
15:47:06.923 -> BLErssi
15:47:06.923 -> -91
15:47:06.958 -> txPower
15:47:06.958 -> -59
15:47:06.958 -> BLE DISTANCE :
15:47:06.958 -> 25.52
15:47:06.958 -> Pub json into:
15:47:06.958 -> home/home_presence/Bedroom
15:47:06.958 -> {"id":"68:2e:ed:07:11:05","manufacturerdata":"⸮","rssi":-91,"distance":25.51913}
15:47:06.958 -> Get services data :
15:47:06.958 -> 1
15:47:08.309 -> Creating BLE buffer
15:47:08.309 -> BLErssi
15:47:08.309 -> -83
15:47:08.309 -> txPower
15:47:08.309 -> 4
15:47:08.309 -> BLE DISTANCE :
15:47:08.309 -> ovf
15:47:08.309 -> Pub json into:
15:47:08.309 -> home/home_presence/Bedroom
15:47:08.309 -> {"id":"74:f6:1c:03:e8:cc","name":"HTC BS CC59BC","rssi":-83,"txpower":4,"distance":1.479725e13}
15:47:15.777 -> BT Task running on core 0
15:47:21.019 -> Creating BLE buffer
15:47:21.019 -> BLErssi
15:47:21.019 -> -86
15:47:21.019 -> txPower
15:47:21.019 -> -59
15:47:21.019 -> BLE DISTANCE :
15:47:21.019 -> 16.55
15:47:21.019 -> Pub json into:
15:47:21.019 -> home/home_presence/Bedroom
15:47:21.019 -> {"id":"76:f1:92:c6:ed:ce","manufacturerdata":"⸮","rssi":-86,"distance":16.54559}
15:47:21.019 -> Get services data :
15:47:21.019 -> 1
15:47:21.019 -> Creating BLE buffer
15:47:21.019 -> BLErssi
15:47:21.019 -> -89
15:47:21.019 -> txPower
15:47:21.019 -> -59
15:47:21.019 -> BLE DISTANCE :
15:47:21.019 -> 21.52
15:47:21.019 -> Pub json into:
15:47:21.019 -> home/home_presence/Bedroom
15:47:21.019 -> {"id":"79:f6:51:f2:ee:95","manufacturerdata":"⸮","rssi":-89,"distance":21.51847}
15:47:21.054 -> Get services data :
15:47:21.054 -> 1
15:47:21.054 -> Creating BLE buffer
15:47:21.054 -> BLErssi
15:47:21.054 -> -65
15:47:21.054 -> txPower
15:47:21.054 -> -59
15:47:21.054 -> BLE DISTANCE :
15:47:21.054 -> 2.01
15:47:21.054 -> Pub json into:
15:47:21.054 -> home/home_presence/Bedroom
15:47:21.054 -> {"id":"45:06:27:58:2b:fd","manufacturerdata":"⸮","rssi":-65,"distance":2.009448}
15:47:21.054 -> Get services data :

```

1technophile, 2019-06-20:
thanks for pointing this, just corrected it in the development branch

1technophile, 2019-06-20:
Should be better with my last commit

maxi1134, 2020-03-08:
Hi there!

I just tried setting up the whitelist again (Added a bunch of Hues bulbs w/ Bluetooth in my house and now it's really getting spammed 😅) But it doesn't seem to work.

I do see this in the logs of the ESP32 when sending the Whitelist command:
`T: Hey I got a callback
N: Received json : {"white-list":["E01126898417"]}`

But then it keeps listing everything it sees. Instead of just E0:11:26:89:84:17

1technophile, 2020-03-08:
Hi,

did you followed the [topic ](https://docs.openmqttgateway.com/use/ble.html#setting-a-white-or-black-list)mentionned in the doc to publish your command?

maxi1134, 2020-03-08:
This is what I published, it seems to be the same:

[image]

1technophile, 2020-03-08:
Your topic should end by "config" instead of "set"
`home/OpenMQTTGateway/commands/MQTTtoBT/config`

ahome-wugu, 2020-05-13:
Hi, I have managed to make white list work, but is there any way to temp disable the already set white list and enable later? e.g. {"white-list":"off"} or {"white-list":"false"}, thanks !

1technophile, 2020-05-13:
Not for the moment, you can submit a feature request if you want