1technophile/OpenMQTTGateway #333: [ESP8266 Witty Cloud][ESP-12E] - is it mandatory to use specific pins ?

shafr, 2018-12-27:
**Describe the bug**
Hello! I'm trying to build my own OpenMqttGateway PCB with bells and whistles.
My instance contains sensors on with the voltage:
* 3.3v: IR-in, IR-out, DHT-11
* 5v: RF-in, RFout.
* ESP8266 Witty Cloud board.

**Actual:**
When everything is connected on the breadboard with the __default pin configuration__ as in default configuration file - it works great. 

As soon as I'm trying to re-wire breadboard and update configuration files to be using different pins - everything falls apart.

I'm getting MQTT online message & version from board, but none of the sensors seems to report any data anymore at once - DHT11 not sending temperature, IR-recv not forwarding received signal to MQTT.

Before uploading every new firmware i'm re-writing my ESP with zeroes, so it's not likely that  issue can be in software.

**Expected:**
Should it be possible to set any sensor to any pin ?  I'm avoiding no-use pins, (GPIO2, RST, CH_PD) but maybe i'm missing something.

**Screenshots**
Just for info the schematic of sensors connection:

[image]

Configuration file specify sensors using GPIO pins.

**Environment (please complete the following information):**
* OpenMQTTGateway version 0.8.1
* I was using Arduino IDE with Generic ESP8266 board & NodeMCU board profile.
* All the required ESP8266 pins are connected - CH_PD, GPIO15, GPIO00,

1technophile, 2018-12-28:
Hello,

If I sum up the constraint for pins selections here is what we have:
* RF or IR receiver should be on an interrupt pin
* HM10 should be on pin that can handle serial connection (not concerning you)
* I2C sensors should be connected to D1 & D2 (not concerning you)
For the rest It should be ok to set any pins not affected to ESP special usage.

shafr, 2018-12-28:
Ok, thank you for the answer! I will continue my investigation.

Can I add this information to a documentation for ESP8266 ? I will do a pull request later with documentation update with pins which were working for me for ESP and a new schematic / PCb.

1technophile, 2018-12-28:
Yes of course, did you seen the pinout spreadsheet ? 
https://docs.google.com/spreadsheets/d/1_5fQjAixzRtepkykmL-3uN3G5bLfQ0zMajM9OBZ1bx0/edit?usp=drivesdk
You can give me your feedbacj and i will update it

saikek, 2019-01-03:
Done:

NodeMCU pin | GPIO PIN | RF Recv | RF Transmitter | IR Recv | IR Transmitter | DHT11
-- | -- | -- | -- | -- | -- | --
D0 | 16 | OK | OK | NOK | OK | NOK
D1 | 5 | OK | OK | OK | OK | OK
D2 | 4 | OK | OK | OK | OK | OK
D3 | 0 | NOK | NOK | NOK | NOK | NOK
D4 | 2 | OK | OK | OK | OK | OK
D55 | 14 | OK | OK | OK | OK | OK
D6 | 12 | OK | OK | OK | OK | OK
D7 | 13 | OK | OK | OK | OK | OK
D8 | 15 | OK | OK | NOK | OK | OK
RX | 3 | OK | OK | OK | OK | OK
TX | 1 | OK | OK | NOK | OK | OK

shafr, 2019-01-03:
NodeMCU pin | GPIO PIN | RF Recv | RF Transmitter | IR Recv | IR Transmitter | DHT11
-- | -- | -- | -- | -- | -- | --
D0 | 16 | OK | OK | NOK | OK | NOK
D1 | 5 | OK | OK | OK | OK | OK
D2 | 4 | OK | OK | OK | OK | OK
D3 | 0 | NOK | NOK | NOK | NOK | NOK
D4 | 2 | OK | OK | OK | OK | OK
D55 | 14 | OK | OK | OK | OK | OK
D6 | 12 | OK | OK | OK | OK | OK
D7 | 13 | OK | OK | OK | OK | OK
D8 | 15 | OK | OK | NOK | OK | OK
RX | 3 | OK | OK | OK | OK | OK
TX | 1 | OK | OK | NOK | OK | OK

1technophile, 2019-01-03:
Wonderfull thanks !

closed 2019-01-03 by shafr