1technophile/OpenMQTTGateway #335: ESP32: No RF Data Send

Patrick-L, 2018-12-30:
Describe the bug
I tried to send RF data via an ESP32. Unfortunately, although the code is recognized correctly and also output via the console, no receiver receives a signal. Receiving signals with a remote control works on the ESP32 however. The setup is as described in the pictures and the send data port is on D12.

To Reproduce:
Install ESP32 with the default config and subscribe it to MQTT. 

Publish Message to MQTT
mosquitto_pub -u XXX --pw XXX -t home/OpenMQTTGateway/433toMQTT -m 1397077 -d -h 192.168.1.210

See error:
```
[14:05:41] patrick@ubuntu-template:~$ sudo mosquitto_sub -t +/# -v -u XXX --pw XXX 
home/OpenMQTTGateway/LWT Online
home/OpenMQTTGateway/version 0.8.1
home/OpenMQTTGateway/433toMQTT 1397077
```

```
Connecting to 
W-Lan-Net
.
.
WiFi ok with manual config credentials
OpenMQTTGateway mac: 
80:7D:3A:FD:C5:C8
OpenMQTTGateway ip: 
192.168.1.210
1883
Connecting to MQTT by IP adress
192.168.1.213
RF_EMITTER_PIN 
12
RF_RECEIVER_PIN 
13
ZgatewayRF setup done 
MQTT_MAX_PACKET_SIZE
1024
Setup OpenMQTTGateway end
MQTT connection...
Connected to broker
Subscription OK to the subjects
Hey I got a callback 
Storing signal
Min ind: 
0
store code :
1397077/72012
Col: val/timestamp
0:1397077/72012
1:0/0
2:0/0
3:0/0
4:0/0
5:0/0
6:0/0
7:0/0
8:0/0
9:0/0
10:0/0
11:0/0
Data stored
```
However, none of the sockets react, regardless of how close the transmitter is. When switching using the remote control, both the power outlet and ESP32 report a signal.

Screenshots
[image]

It is my first published issue, so just let me know, if there is anything missing. 
Thanks! :-)

1technophile, 2018-12-30:
Hello, you are not publishing to the good topic, you should take a look to the RF user guide into the wiki.

Patrick-L, 2018-12-30:
I was so sure that I tried all of those....
But obviously I did not. 
Sorry and thank you! 

How can I close this?

closed 2018-12-30 by Patrick-L