1technophile/OpenMQTTGateway #336: RFM69 issue initialization failed

savage872, 2019-01-02:
I'm using version 0.8 and RFM69HW
[https://www.ebay.co.uk/itm/RFM69HW-868Mhz-433Mhz-915Mhz-20dBm-HopeRF-Wireless-Transceiver-For-Remote-HM-C/272412839767?ssPageName=STRK%3AMEBIDX%3AIT&var=571383421421&_trksid=p2057872.m2749.l2649](url)
Serial connection is showing:
ZgatewayRF setup done 
ZgatewayRFM69 initialization failed
ZgatewayRFM69 Listening and transmitting at
433
I have set the gateway to use 868 but for some reason it's using 433

this is from config_RFM69.h

#define NETWORKID     4  // was 200the same on all nodes that talk to each other
#define NODEID        4 // was 10

//Match frequency to the hardware version of the radio
//#define FREQUENCY     RF69_433MHZ
#define FREQUENCY     RF69_868MHZ
//#define FREQUENCY      RF69_915MHZ
#define IS_RFM69HCW    true // set to 'true' if you are using an RFM69HCW module
#define POWER_LEVEL    31

Any suggestion why it wouldn't initialize?

1technophile, 2019-01-02:
Could you comment zgatewayrf also ?

savage872, 2019-01-02:
I just commented zgatewayrf and uploaded new sketch and it's the same.

1technophile, 2019-01-02:
I would double check the wiring and maybe try at different frequencies.

savage872, 2019-01-02:
I have changed ZgatewayRFM69.ino 

  trc(F("ZgatewayRFM69 Listening and transmitting at"));
  switch (pGC->rfmfrequency) {
//    case RF69_433MHZ:
//      freq = 433;
//      break;
    case RF69_868MHZ:
      freq = 868;
      break;
//    case RF69_915MHZ:
//      freq = 915;
//      break;
//    case RF69_315MHZ:
//      freq = 315;
//      break;
    default:
      freq = -1;
      break;
  }
  trc(freq);

and serial is showing:

ZgatewayIR setup done 
ZgatewayRFM69 Listening and transmitting at
-1

1technophile, 2019-01-02:
Strange, try to replace the -1 by 868 maybe  : 
freq = 868;

savage872, 2019-01-02:
I have commented 
default:
freq = -1;
break; 
and only left 868 and it's now working.
I now need to figure out how to intercept the messages that my heatmiser wireless thermostat is sending to fire the boiler. :) 
Thanks for all your help!

1technophile, 2019-01-02:
Hmm not sure how you will do the interception, if you succeed don't hesitate to post a thread to omg forum.

closed 2019-01-02 by 1technophile