1technophile/OpenMQTTGateway #341: RF 433MHz, one works but not the other one

Beachbuster, 2019-01-08:
Running v.0.8.1
I have 2 different outlets I would like to control, but one of them doesn't work.
Log on the one that doesn't work
[08/01/2019 17:38:05.749] 2 — home/OpenMQTTGateway/433toMQTT/protocol
[08/01/2019 17:38:05.797] 32 — home/OpenMQTTGateway/433toMQTT/bits
[08/01/2019 17:38:05.846] 713 — home/OpenMQTTGateway/433toMQTT/length
[08/01/2019 17:38:05.846] 1758657235 — home/OpenMQTTGateway/433toMQTT

[08/01/2019 17:38:08.704] 2 — home/OpenMQTTGateway/433toMQTT/protocol
[08/01/2019 17:38:08.762] 32 — home/OpenMQTTGateway/433toMQTT/bits
[08/01/2019 17:38:08.762] 713 — home/OpenMQTTGateway/433toMQTT/length
[08/01/2019 17:38:08.762] 4158101651 — home/OpenMQTTGateway/433toMQTT

Command for the one that doesn't work
On
mosquitto_pub -t home/OpenMQTTGateway/commands/PLSL_713/433_2/RFBITS_32 -m 1758657235
Off
mosquitto_pub -t home/OpenMQTTGateway/commands/PLSL_713/433_2/RFBITS_32 -m 4158101651

Command for the one that works
On
mosquitto_pub -t home/OpenMQTTGateway/commands/PLSL_437/433_1/RFBITS_24 -m 1381717
Off
mosquitto_pub -t home/OpenMQTTGateway/commands/PLSL_437/433_1/RFBITS_24 -m 1381716
So receive and transmitt works

Any suggestions what can be wrong?

1technophile, 2019-01-08:
Could you try to command them with the rcswitch examples provided with the library:
https://github.com/1technophile/OpenMQTTGateway/tree/v0.8.1/lib/rc-switch/examples

Try the receivedemo advanced and use the parameter with a send example.

1technophile, 2019-01-08:
Alternatively you can also try ZgatewayPilight with v0.9beta.

Beachbuster, 2019-01-08:
Hi and thanks for very fast reply.
I tested 0.9beta today but it didn't work with both with RF, but I didn't test ZgatewayPilight.
I can test tomorrow again since I'm not sure about how to do it with receivedemo advanced.

Beachbuster, 2019-01-08:
Ok, I've tried v.0.9beta now and with RF it worked same as before (I uncomment simple....)
Now uncomment ZgatewayRF and use only ZgatewayPilight it looks like this:
Command for the one that doesn't work
On
[08/01/2019 22:53:10.885] {"id":3495936,"unit":1,"state":"on"} — home/OpenMQTTGateway/PilighttoMQTT/message
[08/01/2019 22:53:10.954] arctech_switch — home/OpenMQTTGateway/PilighttoMQTT/protocol
[08/01/2019 22:53:10.954] 3495936 — home/OpenMQTTGateway/PilighttoMQTT/length
[08/01/2019 22:53:10.954] 102 — home/OpenMQTTGateway/PilighttoMQTT/repeats
[08/01/2019 22:53:10.954] 2 — home/OpenMQTTGateway/PilighttoMQTT/status
[08/01/2019 22:53:10.954] {"id":3495936,"unit":1,"state":"up"} — home/OpenMQTTGateway/PilighttoMQTT/message
[08/01/2019 22:53:10.954] arctech_screen — home/OpenMQTTGateway/PilighttoMQTT/protocol
[08/01/2019 22:53:10.954] 3495936 — home/OpenMQTTGateway/PilighttoMQTT/length
[08/01/2019 22:53:10.954] 102 — home/OpenMQTTGateway/PilighttoMQTT/repeats
[08/01/2019 22:53:10.954] 2 — home/OpenMQTTGateway/PilighttoMQTT/status
[08/01/2019 22:53:11.001] {"id":3495936,"unit":1,"state":"opened"} — home/OpenMQTTGateway/PilighttoMQTT/message
[08/01/2019 22:53:11.001] arctech_contact — home/OpenMQTTGateway/PilighttoMQTT/protocol
[08/01/2019 22:53:11.001] 3495936 — home/OpenMQTTGateway/PilighttoMQTT/length
[08/01/2019 22:53:11.001] 102 — home/OpenMQTTGateway/PilighttoMQTT/repeats
[08/01/2019 22:53:11.001] 2 — home/OpenMQTTGateway/PilighttoMQTT/status
[08/01/2019 22:53:11.515] {"id":2,"unit":4,"state":"off"} — home/OpenMQTTGateway/PilighttoMQTT/message
[08/01/2019 22:53:11.571] conrad_rsl_switch — home/OpenMQTTGateway/PilighttoMQTT/protocol
[08/01/2019 22:53:11.571] 2 — home/OpenMQTTGateway/PilighttoMQTT/length
[08/01/2019 22:53:11.571] 102 — home/OpenMQTTGateway/PilighttoMQTT/repeats
[08/01/2019 22:53:11.571] 2 — home/OpenMQTTGateway/PilighttoMQTT/status

Off
[08/01/2019 22:53:41.911] {"id":3495936,"unit":1,"state":"off"} — home/OpenMQTTGateway/PilighttoMQTT/message
[08/01/2019 22:53:41.974] arctech_switch — home/OpenMQTTGateway/PilighttoMQTT/protocol
[08/01/2019 22:53:42.027] 3495936 — home/OpenMQTTGateway/PilighttoMQTT/length
[08/01/2019 22:53:42.027] 102 — home/OpenMQTTGateway/PilighttoMQTT/repeats
[08/01/2019 22:53:42.027] 2 — home/OpenMQTTGateway/PilighttoMQTT/status
[08/01/2019 22:53:42.027] {"id":3495936,"unit":1,"state":"down"} — home/OpenMQTTGateway/PilighttoMQTT/message
[08/01/2019 22:53:42.027] arctech_screen — home/OpenMQTTGateway/PilighttoMQTT/protocol
[08/01/2019 22:53:42.027] 3495936 — home/OpenMQTTGateway/PilighttoMQTT/length
[08/01/2019 22:53:42.027] 102 — home/OpenMQTTGateway/PilighttoMQTT/repeats
[08/01/2019 22:53:42.027] 2 — home/OpenMQTTGateway/PilighttoMQTT/status
[08/01/2019 22:53:42.027] {"id":3495936,"unit":1,"state":"closed"} — home/OpenMQTTGateway/PilighttoMQTT/message
[08/01/2019 22:53:42.027] arctech_contact — home/OpenMQTTGateway/PilighttoMQTT/protocol
[08/01/2019 22:53:42.027] 3495936 — home/OpenMQTTGateway/PilighttoMQTT/length
[08/01/2019 22:53:42.027] 102 — home/OpenMQTTGateway/PilighttoMQTT/repeats
[08/01/2019 22:53:42.027] 2 — home/OpenMQTTGateway/PilighttoMQTT/status
[08/01/2019 22:53:42.475] {"id":4,"unit":2,"state":"off"} — home/OpenMQTTGateway/PilighttoMQTT/message
[08/01/2019 22:53:42.531] conrad_rsl_switch — home/OpenMQTTGateway/PilighttoMQTT/protocol
[08/01/2019 22:53:42.531] 4 — home/OpenMQTTGateway/PilighttoMQTT/length
[08/01/2019 22:53:42.531] 102 — home/OpenMQTTGateway/PilighttoMQTT/repeats
[08/01/2019 22:53:42.531] 2 — home/OpenMQTTGateway/PilighttoMQTT/status

Command for the one that works
On
[08/01/2019 22:54:37.531] {"id":7,"state":"off"} — home/OpenMQTTGateway/PilighttoMQTT/message
[08/01/2019 22:54:37.583] selectremote — home/OpenMQTTGateway/PilighttoMQTT/protocol
[08/01/2019 22:54:37.583] 7 — home/OpenMQTTGateway/PilighttoMQTT/length
[08/01/2019 22:54:37.583] 102 — home/OpenMQTTGateway/PilighttoMQTT/repeats
[08/01/2019 22:54:37.583] 2 — home/OpenMQTTGateway/PilighttoMQTT/status
[08/01/2019 22:54:37.583] {"id":29,"unit":14,"state":"on"} — home/OpenMQTTGateway/PilighttoMQTT/message
[08/01/2019 22:54:37.583] arctech_switch_old — home/OpenMQTTGateway/PilighttoMQTT/protocol
[08/01/2019 22:54:37.583] 29 — home/OpenMQTTGateway/PilighttoMQTT/length
[08/01/2019 22:54:37.583] 102 — home/OpenMQTTGateway/PilighttoMQTT/repeats
[08/01/2019 22:54:37.583] 2 — home/OpenMQTTGateway/PilighttoMQTT/status

Off
[08/01/2019 22:54:41.051] {"id":7,"state":"off"} — home/OpenMQTTGateway/PilighttoMQTT/message
[08/01/2019 22:54:41.120] selectremote — home/OpenMQTTGateway/PilighttoMQTT/protocol
[08/01/2019 22:54:41.120] 7 — home/OpenMQTTGateway/PilighttoMQTT/length
[08/01/2019 22:54:41.120] 102 — home/OpenMQTTGateway/PilighttoMQTT/repeats
[08/01/2019 22:54:41.120] 2 — home/OpenMQTTGateway/PilighttoMQTT/status

1technophile, 2019-01-09:
You can now deactivate simplePublishing without impacting the reception (avoiding by the way to pollute your broker), if we sumup we have:
Command for the one that doesn't work
ON [08/01/2019 22:53:11.001] {"id":3495936,"unit":1,"state":"opened"} — home/OpenMQTTGateway/PilighttoMQTT/message
OFF [08/01/2019 22:53:42.027] {"id":3495936,"unit":1,"state":"closed"} — home/OpenMQTTGateway/PilighttoMQTT/message

Command for the one that works
ON [08/01/2019 22:54:37.583] {"id":29,"unit":14,"state":"on"} — home/OpenMQTTGateway/PilighttoMQTT/message
OFF [09/01/2019 07:48:11.227] {"id":29,"unit":14,"state":"off"} — home/OpenMQTTGateway/PilighttoMQTT/message

Even if there is some "unwanted" messages there is a consistency, if you use this [tip ](https://community.openmqttgateway.com/t/correct-message-for-mqtttopilight/257/3?u=1technophile) do you get it working?

1technophile, 2019-01-10:
Could you post what you are seeing on the serial monitor please

Beachbuster, 2019-01-10:
Sure, but I'll train now so I can't do it now.
Unfortunately, I have to do it tomorrow.

Beachbuster, 2019-01-16:
Hi,
Any news you could share with me?
Thanks in advance

1technophile, 2019-01-17:
Hello,

After some tests here is two examples of commands that worked for me:
**ON**
mosquitto_pub -t "home/OpenMQTTGateway_ESP8266/commands/MQTTtoPilight" -m '{"message":"{\"systemcode\":12,\"unitcode\":22,\"on\":1}","protocol":"elro_400_switch"}'
**OFF**
mosquitto_pub -t "home/OpenMQTTGateway_ESP8266/commands/MQTTtoPilight" -m '{"message":"{\"systemcode\":12,\"unitcode\":22,\"off\":1}","protocol":"elro_400_switch"}'

Please note that the state parameter is replaced by on or off with a value equal to 1

Beachbuster, 2019-01-17:
Ciao,
Nope, I tried both down but it didn't work:-) (I have ESP8266 so I did wrong before with ESP32)
mosquitto_pub -t "home/OpenMQTTGateway_ESP8266/commands/MQTTtoPilight" -m '{"message":"{"systemcode":29,"unitcode":14,"on":1}","protocol":"arctech_switch_old"}'

mosquitto_pub -t "home/OpenMQTTGateway_ESP8266/commands/MQTTtoPilight" -m '{"message":"{"systemcode":29,"unitcode":14,"on":1}","protocol":"arctech_switch_old","length":"29","repeats":102,"status":2}'
And
mosquitto_pub -t "home/OpenMQTTGateway_ESP8266/commands/MQTTtoPilight" -m '{"message":"{"systemcode":3495936,"unitcode":1,"on":1}","protocol":"arctech_switch"}'

mosquitto_pub -t "home/OpenMQTTGateway_ESP8266/commands/MQTTtoPilight" -m '{"message":"{"systemcode":3495936,"unitcode":1,"on":1}","protocol":"arctech_switch","length":"3495936","repeats":102,"status":2}'

Beachbuster, 2019-01-26:
Any news about this?
Thanks for your works for this:-)

1technophile, 2019-02-15:
Hello,

Could you test with this format:
https://github.com/1technophile/OpenMQTTGateway/wiki/User-guide-PiLight-09#send-data-by-mqtt-to-convert-it-on-rf-signal

Beachbuster, 2019-03-18:
Ciao,

Sorry, I have been in vacation.
I'll test it this week, let me get back with information.

1technophile, 2019-06-21:
I'm closing this issue, if you need other info don't hesitate to reply

closed 2019-06-21 by 1technophile