1technophile/OpenMQTTGateway #342: Timeout on Home Assistant Mosquitto broker

simonkamronn, 2019-01-08:
I'm using OMG on a Wemos D1 and it seems to work. It connects to WiFi, it connects to the MQTT broker running on HA, and I pick up the RF signals from a Kerui D026 sensor. I have a few ESPURNA devices that have no problems.

The problem is that it continuously disconnects from the MQTT broker from which I get this message:
```
1546979562: New client connected from 192.168.1.153 as OpenMQTTGateway (c1, k5, u'rftomqtt').
1546979569: Client OpenMQTTGateway has exceeded timeout, disconnecting.
1546979569: Socket error on client OpenMQTTGateway, disconnecting.
1546979569: New connection from 192.168.1.153 on port 1883.
[INFO] found rftomqtt on local database
1546979569: New client connected from 192.168.1.153 as OpenMQTTGateway (c1, k5, u'rftomqtt').
1546979583: Client OpenMQTTGateway has exceeded timeout, disconnecting.
1546979583: Socket error on client OpenMQTTGateway, disconnecting.
1546979583: New connection from 192.168.1.153 on port 1883.
[INFO] found rftomqtt on local database
```
From the Serial output I get
```
21:41:11.590 -> MQTT connection...
21:41:11.791 -> Connected to broker
21:41:11.791 -> Subscription OK to the subjects
21:41:18.579 -> MQTT connection...
21:41:18.816 -> Connected to broker
21:41:18.816 -> Subscription OK to the subjects
```
and on the MQTT sub
```
home/OpenMQTTGateway/LWT Offline
home/OpenMQTTGateway/LWT Offline
home/OpenMQTTGateway/LWT Offline
```
I've tried decreasing the MQTT_KEEPALIVE interval to 5 seconds and increasing MQTT_SOCKET_TIMEOUT to 60 but to no avail. Any clues to whats going on?

1technophile, 2019-01-09:
Could you indicate the version:
- version of OMG
- pubsubclient library you are using 
- and the ESP8266 board version from Arduino IDE board manager please

simonkamronn, 2019-01-10:
OMG 8.1
pubsubclient by knolleary in /lib
LOLIN(WEMOS) D1 R2 & mini from https://github.com/esp8266/Arduino

I'm not sure the board version is correct, but it seems to work.

1technophile, 2019-01-10:
So as to know the version of the board in board manager you have to open your arduino IDE, go to boards -> board manager look for esp8266 and  you should see the board version.

simonkamronn, 2019-01-10:
Sorry, didn't know you meant that version number. It's 2.5.0-beta2. You think unstable wifi might be the issue?

1technophile, 2019-01-10:
I'm using and doing my test with 2.4.2, could you try to revert to this one?

simonkamronn, 2019-01-11:
That solved the problem and also helped with too many duplicated messages, thanks. I still have an issue with inconsistent codes from Kerui D026 but that seems to be common.

closed 2019-01-11 by simonkamronn