1technophile/OpenMQTTGateway #344: enable RF433 function and RF315  function  that will lose wireless data

isharrrry, 2019-01-10:
**Describe the bug**
when i  
```
#define ZgatewayRF     "RF"
#define ZgatewayRF315  "RF315"
```
enable RF 433MHz function and RF 315MHz  function  that will not accurate wireless data，the 315MHz value output in RFToMQTT （433MHz）.
i think because both use  RCSwitch library ，when RCSwitch.h file has static variable define .
```
    static int nReceiveTolerance;
    volatile static unsigned long long nReceivedValue;
    volatile static unsigned int nReceivedBitlength;
    volatile static unsigned int nReceivedDelay;
    volatile static unsigned int nReceivedProtocol;
    const static unsigned int nSeparationLimit;
```
when both use available() function that static variable  cloud not accurate tell which object ？mySwitch or mySwitch315？ the RF 433MHz function run first，so the RF 315MHz value output in RFToMQTT ？

**To Reproduce**
  enable :
#define ZgatewayRF     "RF"
#define ZgatewayRF315  "RF315"
 and use 315MHz remote control

**Screenshots**

**Environment (please complete the following information):**
*OpenMQTTGateway version used 0.9b
*RCSwitch  library by lib. dir.  V2.6.2

And ,sorry by my English language bad.

1technophile, 2019-01-10:
Hello,

Did you changed the pins in config_RF.h or are they the same compared to config_RF315.h?

isharrrry, 2019-01-16:
> 你好，
> 
> 您是否更改了config_RF.h中的引脚，或者它们与config_RF315.h相比是否相同？

> Hello,
> 
> Did you changed the pins in config_RF.h or are they the same compared to config_RF315.h?

Yes, I have changed pin , use difference pin or try use default pin connect 315 send , 315 recv. ,433 send and 433 recv. four module .
And I try switching order RFToMQTT() with RF315ToMQTT() ,that will only recv. 315MHZ message .So i think is static variables problem.

isharrrry, 2019-02-16:
Both can't work at the one time

1technophile, 2019-02-16:
Strange I don't see any reason why emitting could not work on both gateways on the same board. I will test it

isharrrry, 2019-03-04:
> Strange I don't see any reason why emitting could not work on both gateways on the same board. I will test it

a simple test. use default code and pin
Connect the RF 433mhz  signal to the pin 0 // #define RF_RECEIVER_PIN 0 // D3 on nodemcu
Press the remote control
message：

```
Creating RF315 buffer
Rcv. RF315
LED MNG
isAduplicate?
Adv data RF315toMQTT
Pub json into:
home/RuiZhuRF/315toMQTT
{"value":3043441,"protocol":1,"length":24,"delay":324}
Store to avoid duplicate
Min ind: 
7
store code :
3043441/855583
Col: val/timestamp
0:23776/541580
1:3043441/546615
2:3043441/549970
3:3043441/553563
4:3043441/557448
5:3043441/575932
6:3043441/580270
7:3043441/855583
8:380430/526281
9:3043441/535932
10:3043441/538949
11:380430/539030
Creating RF315 buffer
Rcv. RF315
LED MNG
isAduplicate?
--don't pub. duplicate--
Creating RF315 buffer
Rcv. RF315
LED MNG
isAduplicate?
--don't pub. duplicate--
```

Only change  signal pin 0 to the pin 4 //#define RF315_RECEIVER_PIN 4 // D3 on nodemcu
Press the remote control
message：

```
Creating RF315 buffer
Rcv. RF315
LED MNG
isAduplicate?
Adv data RF315toMQTT
Pub json into:
home/RuiZhuRF/315toMQTT
{"value":3043441,"protocol":1,"length":24,"delay":323}
Store to avoid duplicate
Min ind: 
8
store code :
3043441/991919
Col: val/timestamp
0:23776/541580
1:3043441/546615
2:3043441/549970
3:3043441/553563
4:3043441/557448
5:3043441/575932
6:3043441/580270
7:3043441/855583
8:3043441/991919
9:3043441/535932
10:3043441/538949
11:380430/539030
Creating RF315 buffer
Rcv. RF315
LED MNG
isAduplicate?
--don't pub. duplicate--

```

As mentioned above, it has been 315 in preemption work, 433 is not working

1technophile, 2019-03-04:
I don't see the link between my remark and the examples given, I'm talking about emitting and you are giving extract of receiving.
When I'm talking about emitting I'm talking about the gateway as an emitter.

isharrrry, 2019-03-04:
> I don't see the link between my remark and the examples given, I'm talking about emitting and you are giving extract of receiving.
> When I'm talking about emitting I'm talking about the gateway as an emitter.

I am sorry, my friend. The gateway as an emitter is not problem，i away talking receiving

1technophile, 2019-05-04:
Currently discussed here:
https://community.openmqttgateway.com/t/is-it-possible-to-have-315-and-433-at-the-same-time/442/6

closed 2019-05-04 by 1technophile