1technophile/OpenMQTTGateway #345: RF433 can received but can't send

TT8174, 2019-01-10:
Hi I'm a beginner,and I already try this project two weeks.
It's hard to me.But I think I will success.
Thanks 1technophile and other  #friends.

Now I can receive signal through SSH. Like:

home/kitchen/433toMQTT {"value":8528372,"protocol":1,"length":24,"delay":221}

But when I pub command.nothing happened. My device not work,and Arduino IDE monitor can't receive RF send singal.
My command is:
mosquitto_pub -t home/OpenMQTTGateway/commands/PLSL_220/433_1/RFBITS_24 -u mqtt -P mqtt -i mosqsub -m 8528372

How can I fix it? Thanks.

1technophile, 2019-01-10:
Hello,

If you download the latest revision of master it should work ootb.

TT8174, 2019-01-10:
In arduino IDE monitor I got this message：

*WM: IP Address:
*WM: 192.168.1.83
connected...yeey :)
saving config
failed to open config file for writing
{"mqtt_server":"192.168.1.101","mqtt_port":"1883","mqtt_user":"mqtt","mqtt_pass":"mqtt"}*WM: freeing allocated params!

So，It's OK?

And in Arduino monitor,I also got this:
IR_EMITTER_PIN 
16
IR_RECEIVER_PIN 
2
ZgatewayIR setup done 
RF_EMITTER_PIN 
3
RF_RECEIVER_PIN 
0
RF setup ok

But my RE_EMITTER_PIN is RX PIN, so It's OK?

Thanks

1technophile, 2019-01-10:
Your issues seems more to be support issue rather than bugs, could you please switch to the community forum 
https://community.openmqttgateway.com

closed 2019-01-10 by 1technophile