1technophile/OpenMQTTGateway #346: valueAsASubject breaks duplication check

PeeVv, 2019-01-10:
When setting valueAsASubject to true, the duplication check breaks and RF -> Mqtt sends multiple messages (+- 30 in 3 seconds in my case) to the broker. I run 0.9beta (built from source) on a NodeMCU v1.0. Setting valueAsASubject to false fixes it, but I would like to use it to prevent the Home Assistant 'No matching payload found for entity' errors.

1technophile, 2019-01-24:
an [enhancement ](https://github.com/1technophile/OpenMQTTGateway/commit/27fa38ec45351bcb4e3c1dc9b4b51b12de19b313)has been apply to duplicate in master, could you check if it correct your issue?

Alternatively please find a [topic ](https://community.home-assistant.io/t/mqtt-py-log-warning-no-matching-payload-found-for-entity/42995/43)discussing about how to avoid the HASS errors

PeeVv, 2019-01-25:
Ah yes, it works now, many thanks!

closed 2019-01-26 by 1technophile