1technophile/OpenMQTTGateway #347: Wifi connectivity issues on my main router since I use OpenMQTTGateway

ejalal, 2019-01-14:
**Describe the bug**
I'm having Wifi connectivity issues on  my main router since I started using OpenMQTTGateway.
My wifi network goes offline for a few minutes multiple times randomly during the day or night it doesn't matter.
I changed router 3 times, I was using a Freebox router as home router, then moved everything (laptops/phones/sensors/OpenMQTTGtw...) to an old Netgear WNR3500v1 router and I was still able to reproduce the issue.
Then I moved the OpenMQTTGateway (ESP32) ALONE to a dedicated Dlink router under DD-WRT so it doesn't disturb the rest of my wifi devices. The issue was solved for my regular home network but the OpenMQTTGateway still disappears behind it's dedicated wifi router. 

**To Reproduce**
Don't know if someone can reproduce, don't know why it happens 

**Expected behavior**
Well, wifi network should not go offline unexpectedly.

**Screenshots**

**Environment (please complete the following information):**
OpenMQTTGateway v0.8.1

**Additional context**

1technophile, 2019-01-17:
Never seen that :-(, could you post your user_config.h without credentials please and point the arduino esp32 env version you are using.

ejalal, 2019-01-20:
hi, this is my User_config.h file (without the mqtt/wifi passwords)
`/*  
  OpenMQTTGateway  - ESP8266 or Arduino program for home automation 

   Act as a wifi or ethernet gateway between your 433mhz/infrared IR signal  and a MQTT broker 
   Send and receiving command by MQTT
 
  This program enables to:
 - receive MQTT data from a topic and send RF 433Mhz signal corresponding to the received MQTT data
 - publish MQTT data to a different topic related to received 433Mhz signal
 - receive MQTT data from a topic and send IR signal corresponding to the received MQTT data
 - publish MQTT data to a different topic related to received IR signal
 - publish MQTT data to a different topic related to BLE devices rssi signal
  
    Copyright: (c)Florian ROBERT
  
    This file is part of OpenMQTTGateway.
    
    OpenMQTTGateway is free software: you can redistribute it and/or modify
    it under the terms of the GNU General Public License as published by
    the Free Software Foundation, either version 3 of the License, or
    (at your option) any later version.

    OpenMQTTGateway is distributed in the hope that it will be useful,
    but WITHOUT ANY WARRANTY; without even the implied warranty of
    MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
    GNU General Public License for more details.

    You should have received a copy of the GNU General Public License
    along with this program.  If not, see <http://www.gnu.org/licenses/>.
*/
/*-------------------VERSION----------------------*/
#define OMG_VERSION "0.8.1"

/*-------------DEFINE YOUR MQTT PARAMETERS BELOW----------------*/
//MQTT Parameters definition
//#define mqtt_server_name "www.mqtt_broker.com" // instead of defining the server by its IP you can define it by its name, uncomment this line and set the correct MQTT server host name
char mqtt_user[20] = "username"; // not compulsory only if your broker needs authentication
char mqtt_pass[20] = "password"; // not compulsory only if your broker needs authentication
char mqtt_server[40] = "192.168.0.IP";
char mqtt_port[6] = "1883";

#define Gateway_Name "OpenMQTTGateway"
#define Base_Topic "home/"
#define version_Topic  Base_Topic Gateway_Name "/version"
#define will_Topic  Base_Topic Gateway_Name "/LWT"
#define will_QoS 0
#define will_Retain true
#define will_Message "Offline"
#define Gateway_AnnouncementMsg "Online"

/*-------------DEFINE YOUR NETWORK PARAMETERS BELOW----------------*/

//#define ESPWifiManualSetup true //uncomment you don't want to use wifimanager for your credential settings on ESP
#define WifiManager_password "your_password"
//#define MDNS_SD //uncomment if you  want to use mdns for discovering automatically your ip server, please note that MDNS with ESP32 can cause the BLE to not work
//#define cleanFS true //uncomment if you want to clean the ESP memory and reenter your credentials
#define maxMQTTretry 4 //maximum MQTT connection attempts before going to wifi setup

//set minimum quality of signal so it ignores AP's under that quality
#define MinimumWifiSignalQuality 8

// Update these with values suitable for your network.
#if defined(ESP32) || defined(ESPWifiManualSetup) // for nodemcu, weemos and esp8266
  #define wifi_ssid "ssid"
  #define wifi_password "password"
#else // for arduino + W5100
  const byte mac[] = {  0xDE, 0xED, 0xBA, 0xFE, 0x54, 0x95 }; //W5100 ethernet shield mac adress
#endif

[... 40 lines cut ...]
/*----------------------------USER PARAMETERS-----------------------------*/
#ifdef ZgatewaySRFB
  #define SERIAL_BAUD 19200
#else
  #define SERIAL_BAUD 115200
#endif
/*--------------MQTT general topics-----------------*/
// global MQTT subject listened by the gateway to execute commands (send RF, IR or others)
#define subjectMQTTtoX  Base_Topic Gateway_Name "/commands/#"
#define subjectMultiGTWKey "toMQTT"

//variables to avoid duplicates
#define time_avoid_duplicate 3000 // if you want to avoid duplicate mqtt message received set this to > 0, the value is the time in milliseconds during which we don't publish duplicates

//uncomment to use multicore function of ESP32 for BLE
#ifdef ESP32
  #define multiCore //comment to don't use multicore function of ESP32 for BLE
#endif

#define TimeBetweenReadingSYS 120000 // time between system readings (like memory)
#define subjectSYStoMQTT  Base_Topic Gateway_Name "/SYStoMQTT"

/*-------------------ACTIVATE TRACES----------------------*/
#define TRACE 1  // 0= trace off 1 = trace on
`

1technophile, 2019-02-15:
Hello,

Do you see something when the serial monitor is connected, like unstable wifi connection?

1technophile, 2019-04-15:
I'm closing this issue, if you have more info you can add them here

closed 2019-04-15 by 1technophile