1technophile/OpenMQTTGateway #351: RF stop on Nodemcu when BLE is activated

1technophile, 2019-01-31:
**Describe the bug**
Reported by @jeanniquini
On NodeMCU with V0.9beta with RF and BLE activated, RF stop after a moment

**To Reproduce**
Steps to reproduce the behavior:
1. Activate ZgatewayBT
2. Activate ZgatewayRF
3. Upload the program to a nodemcu

**Expected behavior**
No crash

**Environment (please complete the following information):**
* OpenMQTTGateway V0.9beta

**Additional context**
@jeanniquini, could you indicate wich version of nodemcu hardware you have, and the ESP8266 version used in arduino IDE (tools->boards->boards manager)

jeanniquini, 2019-02-01:
The whole problem is in miband's BLE. When attempting to extract service data. When it is far away, there are no problems (tested). I am using a separate mcu for BLE and only problem is that the time between the scans is not followed. But when used together with RF, it stops after a while.

jeanniquini, 2019-02-01:
nodemcu v2 Amica

[image]

1technophile, 2019-02-03:
Please click on Placa "NodeMCU 1.0 ...", on board manager, search for esp and you should have the ESP8266 board version (2.5.0beta or 2.4.2 or other)

jeanniquini, 2019-02-03:
2.4.2

1technophile, 2019-02-15:
Could you indicate about how long does it take to crash?

jeanniquini, 2019-02-16:
@1technophile I have tested many times. There is no clear time. It depends on how much time Mi Band is around. The shortest time until RF stop was around 3 hours approximately.

jeanniquini, 2019-02-25:
@1technophile 
`Mi band 3

Creating BLE buffer
Pub json into:
home/OpenMQTTGateway_BLE/BTtoMQTT/CDEDE7B9F4D2
{"rssi":-51}
4f4b2b444953413ad2f4b9e7edcd01cd1f0201061bff57010036eae16345127ea5e151aea4c712cf4203cdede7b9f4d20d0a
Creating BLE buffer
Pub json into:
home/OpenMQTTGateway_BLE/BTtoMQTT/CDEDE7B9F4D2
{"rssi":-51}
4f4b2b444953413a98396acb696d03d50a09ffe0001640ca7ec6480d0a
4f4b2b444953413a98396acb696d03d51c03039ffe17169ffe00000000000000000000000000000000000000000d0a

Google Home

Creating BLE buffer
Pub json into:
home/OpenMQTTGateway_BLE/BTtoMQTT/6D69CB6A3998
{"rssi":-45}
`

closed 2019-08-14 by stale[bot]