1technophile/OpenMQTTGateway #354: MQTT server settings are not saved in the WIFI manager.

odis68, 2019-02-03:
version 0.9. MQTT server settings are not saved in the WIFI manager. ESP8266 After a power failure, the settings return to 192.168.1.17. On version 0.8 this was not observed.

1technophile, 2019-02-03:
does it pop up the web page for setting the credential also?

odis68, 2019-02-03:
Yes, where is the installation of the WiFi ssid address MQTT login and password. After installation, everything connects to WiFi and the broker and then how you take off the power, everything disappears. It is interesting that when you returned to version 0.8 it was not necessary to enter the settings, restore yourself and everything connected.

1technophile, 2019-02-03:
the configuration for mqtt is [here](https://github.com/1technophile/OpenMQTTGateway/blob/0180a0dbd55ed8e0799e30ee84f68070a6f478fa/User_config.h#L69)

odis68, 2019-02-03:
I take the finished firmware. bin file

1technophile, 2019-02-04:
Thanks for the info I will test it.

1technophile, 2019-02-05:
Here is my results:
-upload from Arduino IDE -> I don't reproduce the issue
-upload from esptool with the bin file present in the v0.9 release page-> There is an error when mounting the file system of the ESP

I have rebuilt the bin files,
flashed [this one](https://github.com/1technophile/OpenMQTTGateway/releases/download/v0.9/ESP8266_OpenMQTTGateway_V09_IRRFBT.ino.bin.zip)

And now there is no error of mounting FS anymore and by the way the parameters are conserved.

Please note that I have added to the release page several configuration for esp8266 (rf only, rf + ir and rf + ir + bt)

odis68, 2019-02-06:
Thank you so much I will test as soon as I can

1technophile, 2019-02-15:
I close thie issue, if you reproduce it don't hesitate to reopen

closed 2019-02-15 by 1technophile